zephyrproject-rtos/west #21: NameError: global name 'PermissionError' is not defined

nashif, 2018-09-03:
```
nuc01:~: west init /tmp/xxx
('Initializing in new directory', '/tmp/xxx')
Traceback (most recent call last):
  File "/usr/bin/west", line 11, in <module>
    load_entry_point('west==0.2.5', 'console_scripts', 'west')()
  File "build/bdist.linux-x86_64/egg/bootstrap/main.py", line 190, in main
  File "build/bdist.linux-x86_64/egg/bootstrap/main.py", line 127, in init
  File "build/bdist.linux-x86_64/egg/bootstrap/main.py", line 136, in init_bootstrap
NameError: global name 'PermissionError' is not defined
```

nashif, 2018-09-03:
that too, python 2...

mbolivar, 2018-09-06:
> that too, python 2...

This is definitely a Python 2 issue. Python 3 is when PermissionError was added.

closed 2018-09-06 by mbolivar