zephyrproject-rtos/west #22: SyntaxError: invalid syntax when calling west

nashif, 2018-09-03:
```
nuc01:~: cd multirepo/
nuc01:multirepo: ls
manifest  west
nuc01:multirepo: ls -la
total 0
drwxrwxr-x. 1 nashif nashif  34 Sep  3 14:36 .
drwx------. 1 nashif nashif 690 Sep  3 14:36 ..
drwxrwxr-x. 1 nashif nashif  50 Sep  3 14:36 manifest
drwxrwxr-x. 1 nashif nashif   0 Sep  3 14:36 .west
drwxrwxr-x. 1 nashif nashif 216 Sep  3 14:36 west
nuc01:multirepo: west
Traceback (most recent call last):
  File "/usr/bin/west", line 11, in <module>
    load_entry_point('west==0.2.5', 'console_scripts', 'west')()
  File "build/bdist.linux-x86_64/egg/bootstrap/main.py", line 188, in main
  File "build/bdist.linux-x86_64/egg/bootstrap/main.py", line 173, in wrap
  File "/usr/lib64/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/nashif/multirepo/west/src/west/main.py", line 15, in <module>
    from . import log
  File "/home/nashif/multirepo/west/src/west/log.py", line 33
    def dbg(*args, level=VERBOSE_NORMAL):
                       ^
SyntaxError: invalid syntax
```

ulfalizer, 2018-09-03:
Looks like Python 2 is being used. Maybe west was installed with `pip` instead of `pip3`.

nashif, 2018-09-03:
yeah, could be that I mixed p2 with p3, let me check.

carlescufi, 2018-09-04:
@ulfalizer @mbolivar can we protect ourselves against this by detecting it early in the bootstrap code and failing gracefully with something like "west requires Python 3, and it is currently being run by a Python 2 installation"

ulfalizer, 2018-09-04:
Could just make the PyPI wheels Python 3-only. Think it happens automatically if you build them with Python 3. I know you have to explicitly declare if you support both Python 2 and Python 3 at least.

Can tell from the names of the generated wheels which version is supported (get `py3` or similar in the filename).

ulfalizer, 2018-09-04:
For the local installation case, `setup.py` could do something like this to bail out on Python 2:

    if sys.version_info[0] < 3:
        sys.exit("West requires Python 3")

ulfalizer, 2018-09-04:
Hmm... there's already a `python_requires='>=3.4'` in `setup.py` though. Guess it gets ignored by `pip install -e`.

ulfalizer, 2018-09-04:
Or wait... it's getting respected, so scrap that.

Not sure how you ended up with a Python 2 install in that case.

mbolivar, 2018-09-06:
> Not sure how you ended up with a Python 2 install in that case.

I'm guessing this was installed manually with python 2. It will be impossible to install west via pip2 once it's uploaded, since as you pointed out, setup.py prevents that.

(There is a west up on test pypi that will try to install onto python 2, but I'm assuming that wasn't what was used.)

mbolivar, 2018-09-06:
If it still breaks with https://github.com/zephyrproject-rtos/west/pull/20 merged, then it's not Python 2.

mbolivar, 2018-09-21:
@nashif as various other people have had luck with this, I'm guessing it was python 2 -- could you reopen if you still have a problem?

closed 2018-09-21 by mbolivar