zephyrproject-rtos/west #23: unresolved PyYAML dependencies

nashif, 2018-09-03:
not sure what to do about that...

```
nuc01:multirepo: west
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 570, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 888, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 779, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (PyYAML 3.12 (/usr/lib64/python3.6/site-packages), Requirement.parse('PyYAML==3.11'), {'pykwalify'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/nashif/.local/bin/west", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3095, in <module>
    @_call_aside
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3079, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3108, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 572, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 585, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python3.6/site-packages/pkg_resources/__init__.py", line 779, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (PyYAML 3.12 (/usr/lib64/python3.6/site-packages), Requirement.parse('PyYAML==3.11'), {'pykwalify'})

```

ulfalizer, 2018-09-04:
What command did you run when you go that?

Wonder if it's trying to say that PyYAML 3.12 is installed but pykwalify isn't, and pykwalify wants `PyYAML 3.11`, for whatever reason. Can't reproduce it at least.

What versions does `pip3 list` list for PyYAML and pykwalify?

mbolivar, 2018-10-04:
I'm going to close this as a Python 2 problem given that multiple people have run west without this issue across all supported platforms. @nashif please reopen if you continue to have problems.

closed 2018-10-04 by mbolivar

KartikAiyer, 2021-12-03:
@mbolivar  I'm running into this issue when trying `west debug`. Specifically it is 
```
raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (PyYAML 5.4.1 (/home/kt/.local/lib/python3.8/site-packages), Requirement.parse('pyyaml<7.0,>=6.0'), {'cmsis-pack-manager'})
```

I'm using a docker which installs python packages using [requirements.txt](https://raw.githubusercontent.com/zephyrproject-rtos/zephyr/master/scripts/requirements.txt) file from the zephyr project. Looking specifically at [requirements-base.txt](https://github.com/zephyrproject-rtos/zephyr/blob/6a8326ff1e76890af1483ed1ac5ebaa6f80a2d28/scripts/requirements-base.txt#L11) It look like they expectation is PyYAML 5.1 but the exception seems to be looking for PyYAML 7.0. Its not clear if this is a catastrophic error  ? My debugger is not connecting to my board but I'm not sure if its because this exception. 

Any help here would be appreciated.

mbolivar-nordic, 2021-12-04:
Hi, west's setup.py file only requires `PyYAML>=5.1`. A larger version should be fine. I'm not sure where your pip issues are coming from, sorry.

KartikAiyer, 2021-12-04:
HI @mbolivar thanks for you comment. Last question. The error is being reported by `west debug`. Here is the full error 
It wasn't pip that was complaining. Do you know what could be generating that requirement of >6.0 and <7.1 for PyYAML ? To me it looks like west. 

```
kt@c2a5c55913ea:~/src$ west debug                                                                                       
-- west debug: rebuilding                                                                                               
ninja: no work to do.                                                                                                   
-- west debug: using runner pyocd                                                                                       
-- runners.pyocd: pyOCD GDB server running on port 3333                                                                 
GNU gdb (crosstool-NG 1.24.0.378_e011758) 9.2                                                                           
Copyright (C) 2020 Free Software Foundation, Inc.                                                                       
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>                                           
This is free software: you are free to change and redistribute it.                                                      
There is NO WARRANTY, to the extent permitted by law.                                                                   
Type "show copying" and "show warranty" for details.                                                                    
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arm-zephyr-eabi".                                 
Type "show configuration" for configuration details.                                                                    
For bug reporting instructions, please see:                                                                             
<http://www.gnu.org/software/gdb/bugs/>.                                                                                
Find the GDB manual and other documentation resources online at:                                                        
    <http://www.gnu.org/software/gdb/documentation/>.                                                                   
                                                                                                                        
For help, type "help".                                                                                                  
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/kt/src/build/zephyr/zephyr.elf...
Traceback (most recent call last):                     
  File "/home/kt/.local/bin/pyocd", line 5, in <module>
    from pyocd.__main__ import main
  File "/home/kt/.local/lib/python3.8/site-packages/pyocd/__init__.py", line 21, in <module>
    from . import gdbserver
  File "/home/kt/.local/lib/python3.8/site-packages/pyocd/gdbserver/__init__.py", line 17, in <module>
    from .gdbserver import GDBServer
  File "/home/kt/.local/lib/python3.8/site-packages/pyocd/gdbserver/gdbserver.py", line 40, in <module>
    from ..rtos import RTOS
  File "/home/kt/.local/lib/python3.8/site-packages/pyocd/rtos/__init__.py", line 30, in <module>
    load_plugin_classes_of_type('pyocd.rtos', RTOS, ThreadProvider)
  File "/home/kt/.local/lib/python3.8/site-packages/pyocd/core/plugin.py", line 88, in load_plugin_classes_of_type
    plugin = entry_point.load()()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2444, in load
    self.require(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2467, in require
    items = working_set.resolve(reqs, env, installer, extras=self.extras)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 792, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (PyYAML 5.4.1 (/home/kt/.local/lib/python3.8/site-packages), Requirement.parse(
'pyyaml<7.0,>=6.0'), {'cmsis-pack-manager'})
```

marc-hb, 2021-12-04:
> Do you know what could be generating that requirement of >6.0 and <7.1 for PyYAML ? To me it looks like west.
> ...
> pkg_resources.ContextualVersionConflict: (PyYAML 5.4.1 (/home/kt/.local/lib/python3.8/site-packages), Requirement.parse('pyyaml<7.0,>=6.0'), {'cmsis-pack-manager'})

That really didn't take long to find:

https://github.com/pyocd/cmsis-pack-manager/blob/65b3e3eb5/setup.cfg#L40

KartikAiyer, 2021-12-06:
@marc-hb thanks for tracking that down and apologies for not finding it myself. It hadn't occurred to me to look specifically at pyocd.