zephyrproject-rtos/west #27: west and python3.5

tejlmand, 2018-09-12:
When running python3.5, I get the foolowing error when trying to do

`pip3 install west`
`Traceback (most recent call last):`
`  File "/usr/bin/pip3", line 9, in <module>`
`    from pip import main`
`ImportError: cannot import name 'main'`

Running with python3.6 works great.

mbolivar, 2018-09-14:
Thanks for the report!

Based on your error output, it looks like this is an issue in your environment due to a failed pip upgrade at some point in the past:

https://github.com/pypa/pip/issues/5599

Does any of the debugging advice in that issue help?

tejlmand, 2018-09-20:
@mbolivar you were right.
Never seen this issue before, but interesting to know they discourage updating pip by itself when using a distro version (in my case ubuntu).
Have updated pip often, by itself, and never experienced this.

So everything is working now.

closed 2018-09-20 by tejlmand