zephyrproject-rtos/west #31: West fetch looks for manifest in wrong folder

tejlmand, 2018-09-13:
Sep 12, the west init and west fetch worked as expected.
But from Sep 13, it appears west is looking for manifest in wrong location:

`/projects/github/zephyr-west-test$ west init`
`Initializing in /projects/github/zephyr-west-test`
`Cloning into '/projects/github/zephyr-west-test/manifest'...`
`remote: Counting objects: 13, done.`
`remote: Total 13 (delta 0), reused 0 (delta 0), pack-reused 13`
`Unpacking objects: 100% (13/13), done.`
`Checking connectivity... done.`
`Cloning into '/projects/github/zephyr-west-test/west'...`
`remote: Counting objects: 384, done.`
`remote: Compressing objects: 100% (87/87), done.`
`remote: Total 384 (delta 59), reused 58 (delta 26), pack-reused 270`
`Receiving objects: 100% (384/384), 128.42 KiB | 0 bytes/s, done.`
`Resolving deltas: 100% (199/199), done.`
`Checking connectivity... done.`
`/projects/github/zephyr-west-test$ west fetch`
`WARNING: --zephyr-base missing and no ZEPHYR_BASE in the environment`
`FATAL ERROR: Provided source_file do not exists on disk: /projects/github/zephyr-west-test/west/manifest/default.yml`
`run as "west -v ... fetch ..." for a stack trace`
`ubuntu:/projects/github/zephyr-west-test$ `

it looks for the manifest in:
`/projects/github/zephyr-west-test/west/manifest/default.yml `

where the correct location is:
`/projects/github/zephyr-west-test/manifest/default.yml `

Note: `west  fetch -m manifest/default.yml`
is working

ulfalizer, 2018-09-14:
@tejlmand 
This is due to https://github.com/zephyrproject-rtos/west/pull/26, so it's expected. Should be more careful about making changes like that later, once the repo stuff is official.

tejlmand, 2018-09-14:
@ulfalizer 
Yes, but my point is that `west` should check out the manifest in the west folder.

Right now, `west init` clones the manifest to the wrong folder.
But I guess you are already working on this

mbolivar, 2018-09-14:
> Not sure if the latest version has been pushed out to PyPI yet.

It hasn't, we are waiting to tag for self update to work.

Since there are only 4 active users, maybe reorganizing the repository structure and updating west manually is the answer for now?

mbolivar, 2018-09-26:
@tejlmand: as Ulf explained, there was a change in the repository structure (https://github.com/zephyrproject-rtos/west/commit/b7670c75ff98f303aa1fd5d246b2484fa3317194) which was incompatible with the west bootstrapper 0.1.0. I'm therefore closing this issue as expected behavior / early-days-churn.

closed 2018-09-26 by mbolivar