zephyrproject-rtos/west #34: Not possible to checkout remote branch using west.

tejlmand, 2018-09-14:
west tool fails when trying to checkout remote branch using:
`west checkout v1.13-branch zephyr`
fails with error:
`FATAL ERROR: No branch v1.13-branch exists in any of the listed projects`

This is done after `west init` and `west fetch`

Note: Doing this manually with git of course works:
`cd zephyr/`
`zephyr$ git checkout v1.13-branch`
`Branch v1.13-branch set up to track remote branch v1.13-branch from origin.`

ulfalizer, 2018-09-14:
Going to bed, so I'll just dump the IRC discussion as reference for later for now:

```
11:10 < tejlmand> let's say I do `west init` and then `west fetch`, then I end up with zephyr:master
11:11 < tejlmand> I now want to look at the `v1.13-branch`
11:11 < tejlmand> what is the expected workflow here?
11:12 < tejlmand> `west branch` seems to create a new branch
11:13 < tejlmand> `west checkout v1.13-branch zephyr` gives me: `FATAL ERROR: No branch v1.13-branch exists ...`
11:16 < carlesc> tejlmand: in this particular case, where v1.13-branch only exists in Zephyr and not the other 
                 repos (like the manifest) I assume you have to go and check out the revision manually with `git` 
                 inside `zephyr`
11:16 < carlesc> but it's a good point, all of this is open for discussion
11:16 < carlesc> Ulfalizer: martib: thoughts?
11:17 < tejlmand> That's exactly why I specify the checkout only should happen on project `zephyr`
11:18 < tejlmand> From help text:
11:19 < tejlmand> https://www.irccloud.com/pastebin/tM9dKUhT/
11:27 < carlesc> tejlmand: hmm strange, then this is a bug
11:29 < carlesc> tejlmand: can you file a bug?
11:30 < tejlmand> on going
11:42 < Ulfalizer> tejlmand: 'west checkout' is for local branches
11:44 < tejlmand> Ulfalizer: so we don't expect west to be able to checkout remote branch ?
11:44 < Ulfalizer> 'git checkout v1.13-branch' in git is a shorthand for 'git checkout remotes/origin/v1.13' that 
                   works when the name is unique
11:44 < Ulfalizer> tejlmand: was never anything i imagined people wanting to do tbh
11:44 < Ulfalizer> i don't know people's workflows though :)
11:45 < tejlmand> Ulfalizer: what if I would like to checkout v1.13 and then create a new branch, let's 
                  v1.13-local_work
11:45 < tejlmand> Ulfalizer: then I would need to first use `git ` inside zephyr to get the right branch
11:45 < carlesc> ah right, but the thing is that git treats exsiting remote branches as checkoutable branches
11:45 < tejlmand> and then from there use west to create the local branch
11:45 < carlesc> i.e. I often do things like `git checkout <remote-branch-i-have-fetched>
11:46 < Ulfalizer> yup, if you're creating a topic branch for doing some multirepo work that you're gonna submit 
                   later
11:47 < carlesc> Ulfalizer: but say I want to try out the latest Zephyr release
11:47 < Ulfalizer> dunno how much git functionality we should replicate in west. it's meant to solve the stuff 
                   that would be tricky with git alone.
11:47 < carlesc> I should be able to do that with `west` imho
11:47 < Ulfalizer> feature creep can be dangerous
11:47 < carlesc> I agree, but this is kinda basic isn' t it?
11:48 < tejlmand> but I think `checkout` should work similar
11:48 < Ulfalizer> what does repo do there? need to find something that fits in the overall design too.
11:48 < tejlmand> if I would have to use `git` for checking out a remote branch
11:48 < carlesc> Ulfalizer: will `west checkout remotes/origin/v1.13` work with west?
11:48 < Ulfalizer> dumping random feature requests in quickly -> spaghetti land down the line, i think :/
11:48 < tejlmand> then I might just do the rest with git as well
11:49 < carlesc> I think we need to be very careful of what we introduce
11:49 < tejlmand> carlesc: tried that
11:49 < carlesc> but I would like to be able to check out a particular branch that exists remotely on all repos at once with west
11:49 < carlesc> don't you agree Ulfalizer ?
11:49 < tejlmand> carlesc: same error
11:49 < Ulfalizer> carlesc: nopes, it's just local branches now
11:49 < carlesc> if it only exists on one repo... then it's debatable
11:49 < tejlmand> we still have the `-b` flag
11:49 < Ulfalizer> for remote branches, you'd get a detached head. maybe people like that though...
11:50 < carlesc> but if it exists on all repos, I am not convinced
11:50 < tejlmand> to tell the branch should be created
11:50 < carlesc> Ulfalizer: I think detached HEADs are very unpopular, judging by our conversations over time
11:50 < carlesc> tejlmand: please create the issue and let's gather feedback there
11:50 < Ulfalizer> no way around it there at least
11:51 < carlesc> I will also add comments once I try it myself
11:51 < Ulfalizer> it's how it works in plain git
11:51 < tejlmand> carlesc: already did
11:51 < Ulfalizer> tejlmand: think the right thinking is "what's the absolutely most minimal thing we can 
                   introduce that will cover this feature and possible others"?
11:51 < carlesc> Ulfalizer: assuming a remote branch exists on all repos, you can use `git` to just checkout a 
                 local branch with the remote unique branch name right?
11:51 < tejlmand> https://github.com/zephyrproject-rtos/west/issues/34
11:51 < Ulfalizer> to many random one-offs will be a mess
11:51 < Ulfalizer> *too
11:52 < carlesc> Ulfalizer: agreed, that's why I'm trying to limit the scop
[... 85 lines cut ...]
12:13 < Ulfalizer> it's meant to work with those "work branches"
12:13 < Ulfalizer> simple to explain and understand
12:14 < Ulfalizer> think the thing you want might be a better fit for 'forall'
12:14 < Ulfalizer> not against helpers in general, but this one kinda breaks the model i think. maybe something else could be added.
12:16 < carlesc> Ulfalizer: I do see your point. Let's come to a conclusion in the GH issue
12:16 < carlesc> I am not convinced myself one way or the other at this point
[... cut ...]
12:53 < Ulfalizer> git fetch/pull -b FIX-FOO  could fetch/pull all repositories that have that branch
12:54 < Ulfalizer> maybe some general 'for all repositories with this topic branch' thingy would be nice at some point
12:54 < Ulfalizer> urr, *west fetch/pull
12:55 < carlesc> Ulfalizer: interesting, sounds like a good option. Can you add it in the GH issue?
12:55 < Ulfalizer> yeah, not committing to it though. the first version of everything is crap. :)
12:55 < Ulfalizer> something along those lines at least
```

ulfalizer, 2018-09-15:
Note to self: Update documentation to make it clear that local branches are meant.

Some kind of helper could probably be added, if it's something that turns out to be awkward to do in plain Git. Just gotta make sure it fits in.

mbolivar, 2018-09-27:
I think this is a bug -- the idea being "west checkout <branch> <just-one-repository>" should behave consistently with git. Would you agree @ulfalizer? We can set this back to enhancement if not.

carlescufi, 2018-09-28:
I think we can handle this in a clean, non-intrusive way that makes it very obvious to the user what is goin on under the hood:

```
$ west checkout new-feature-from-john
< internally west does a foreach(git checkout new-feature-from-john). If git returns error, stop there>
```

PR #86 (2018-10-26): commands: project: Clarify that 'west checkout' deals with local branches
As long as that's the case, it should be documented.

I suspect it might be a feature. If you've created a branch do to some
work with 'west branch fix-stuff repo1 repo2 repo3', then it's probably
unexpected if 'west checkout fix-stuff' checks out a remote branch
'fix-stuff' that happens to exist in some other repository.

Fixes: #34
commit eee4b239 (2018-10-26): commands: project: Clarify that 'west checkout' deals with local branches

As long as that's the case, it should be documented.

I suspect it might be a feature. If you've created a branch do to some
work with 'west branch fix-stuff repo1 repo2 repo3', then it's probably
unexpected if 'west checkout fix-stuff' checks out a remote branch
'fix-stuff' that happens to exist in some other repository.

Fixes: #34
closed 2018-10-26 by carlescufi