zephyrproject-rtos/west #39: List tags of remote manifest using west

tejlmand, 2018-09-21:
`west` is the future starting point when working with Zephyr and projects based on Zephyr.

It is intended both for skilled developers to facilitate working with features crossing multiple repos, so that PR's can easier be managed.

But `west` is also the starting point for less experienced developers or developers unfamiliar with `git`.
And an clear way of relating multiple repo.

To initialize `west` and fetch latest `zephyr`, one will do:
```
west init
west fetch
```
and if one wants a dedicated revision, the following can be used:
```
west init --mr <revision>
```

However, if user want to see available revision on the manifest, he would need to write:
```
git ls-remote --tags --refs git ls-remote --tags --refs https://github.com/zephyrproject-rtos/zephyr
```

What would be nice, is to tag the manifest when a new release is made, e.g. v1.14, v1.15, etc on the manifest.

Then documentation can be:
```
Using zephyr latest version:
west init
west fetch

To see list of Zephyr releases, and fetch a given version:
west init --lr   <-- Other name can be proposed, functionaliy is important
west init --mr <revision>
west fetch
```

This gives a user a clear ways to find what releases are available, it makes it easier for Zephyr and companies forking Zephyr to distribute multiple releases though the manifest.

And as the process is very clear, it is also very easy to describe.

tejlmand, 2018-09-27:
> >    What would be nice, is to tag the manifest when a new release is made, e.g. v1.14, v1.15, etc on the manifest.
> 
> I definitely agree that this is what must be done to keep everybody sane +1.
>
> This is what the Android team does with their repo manifests: tag the manifests themselves when a new version is released. The tagged manifest ref has projects whose revisions are also tagged to the right place. Here is an example for the android-9.0.0_r8 tag:
> 
> https://android.googlesource.com/platform/manifest/+/android-9.0.0_r8/default.xml
> 
> Notice this line:
>
> <default revision="refs/tags/android-9.0.0_r8"

And that is already possible in `west` 

We can do:
```
   projects:
    - name: zephyr
      revision: zephyr-v1.12.0
...
```
and tag that manifest with zephyr-v1.12.0
Then user runs:
`west init --mr zephyr-v1.12.0`

So that feature is already in place.

> I would like to propose an alternative to adding additional features to the bootstrapper: simply documenting the available tags. See how Android does this in step 3 of their own getting started:
>
> https://source.android.com/setup/build/downloading#initializing-a-repo-client
> 
> The list of tags and what they correspond to is very clear:
> 
> https://source.android.com/setup/start/build-numbers.html#source-code-tags-and-builds

Of course that  should also be available.
Similar to what we have here:
https://docs.zephyrproject.org/latest/release-notes.html#zephyr-release-notes

But as you notice in Android, there is a long list.
And Zephyr list will also grow longer over time.

In case I just want to see if there is something new, it could be nice to do something as:
```
west show releases | grep android-8
```
And get all 8.x related versions.

Then I won't have to go to a web-page, because I know the tool I will be using to fetch the release can also present the list of releases to me.

mbolivar, 2018-09-27:
> So that feature is already in place.

Yup! I designed the manifest schema with this use case in mind.

> west show releases | grep android-8

This type of command would not involve a change to the bootstrapper, which was my concern. I don't get why that's so much easier than `cd west/manifest; git tag | grep android-8`, so I still think it's a bit of feature creep, but I wouldn't object to it.

mbolivar, 2018-09-27:
> I still think it's a bit of feature creep, but I wouldn't object to it.

I guess I would strengthen this to "I think this is reinventing the wheel, and we try hard not to do that in west. But if it really makes your life a lot easier for some customers / users, I wouldn't object to it."

tejlmand, 2018-09-27:
> >    west show releases | grep android-8
> This type of command would not involve a change to the bootstrapper, which was my concern. I don't get why that's so much easier than `cd west/manifest; git tag | grep android-8`

The main issue with your example, is that it expect the repo to already be cloned.
But that repo is not cloned until `west init --mr <revision>` is executed.
Thus, you cannot do the `git tag`, but would need to do:
```
git ls-remote --tags --refs https://github.com/zephyrproject-rtos/zephyr
```

Therefore the `west show ... ` would need to be possible prior to `west init` to show the available tags.

Thinking a bit more about it, there could be another way to solve this.
Moving/adding the `--mr <revision>` argument to `west fetch`, so that one can do:
```
west init
cd west/manifest
git tag | grep <whatever>
cd ../..
west fetch --mr <revision>
```
in which case `west` will use the manifest revision specified by `--mr`.

mbolivar, 2018-09-27:
> The main issue with your example, is that it expect the repo to already be cloned.
> But that repo is not cloned until `west init --mr <revision>` is executed.
> Thus, you cannot do the `git tag`, but [...]

Sorry; I am confused. The idea for `west show` originally came from a comment by you: https://github.com/zephyrproject-rtos/west/issues/39#issuecomment-425006789

I was assuming that you were proposing to move this functionality into a separate west command, as the original proposal to put it in the bootstrapper by modifying `init` was one of my main concerns -- I thought that was your implication by describing this feature in terms of a (new) `show`, i.e. non-`init`, command. Sorry for not verifying that with you first. I remain opposed to extending `init` for this feature, FWIW. The bootstrapper is much harder to change.

> git ls-remote --tags --refs https://github.com/zephyrproject-rtos/zephyr

Hrm, I thought we had agreed earlier that the tags were going to be in the manifest repository. The above indicates the Zephyr repository. Do you still agree that the release tags belong in the manifest repository?

> ```
> west fetch --mr <revision>
> ```
> in which case `west` will use the manifest revision specified by `--mr`.

Or just `cd west/manifest; git checkout <revision>; west fetch`, no?

I am further confused because the above is for checking out a manifest revision, not getting a list of tags in the manifest, which is what this issue is about.

One of the differences between west and repo that has been discussed at length is that in west, the manifest is a first class citizen which is not buried under some hidden directory. We want people to feel free to edit it themselves.

In my experience with repo, editing the manifest directly with git and then executing repo commands provides a very consistent mental model. Wrapping it with additional layers that already have equivalents seems like the wrong direction.

tejlmand, 2018-09-28:
Sorry for the confusion.
The original, `west init --lr` example was simply to describe the functionality.
Whether it is placed as a `west show` or argument to `west init` is less important (to me).
Hence the reason I wrote
> `west init --lr   <-- Other name can be proposed, functionality is important`

I was trying to described the functionality needed, not how the command should look like.

I agree, that the `west init --ls` is not the best place.
`west show` seems to be better.

> Sorry for not verifying that with you first. I remain opposed to extending init for this feature, FWIW. 

Guess I'm the root cause for the confusion here :-s

> >  `git ls-remote --tags --refs https://github.com/zephyrproject-rtos/zephyr`
>
> Hrm, I thought we had agreed earlier that the tags were going to be in the manifest repository. The above indicates the Zephyr repository. Do you still agree that the release tags belong in the manifest repository?

Yes of course
The `git` command example is just to show what kind of `git` command users would need to execute without this feature.
Zephyr was chosen in the example, because
`git ls-remote --tags --refs https://github.com/zephyrproject-rtos/manifest`
currently has no tags on it, and hence returns nothing.

> Edit: of course they also belong in all the others -- my point is that the tagged manifest is the central point of truth for the revisions of all the other projects with that same tag, as explained by analogy with Android above.

Agree

tejlmand, 2018-09-28:
> Or just `cd west/manifest; git checkout <revision>; west fetch`, no?

wouldn't  you need to then specify:
`west fetch --no-update` in that case ?

As the help text says: 
```
$ west fetch -h
usage: west fetch [-h] [-m MANIFEST] [--no-update] [PROJECT [PROJECT ...]]

Clone/fetch projects.

Fetches upstream changes in each of the specified projects (default:
all projects). Repositories that do not already exist are cloned.

Unless --no-update is passed, the manifest and West source code
repositories are updated prior to fetching. See the 'update' command.
```

mbolivar, 2018-09-28:
> wouldn't you need to then specify:
> `west fetch --no-update` in that case ?

We were thinking the same thing!

I realized last night one of the differences between our update behavior and repo's is that repo will only automatically update the manifest if you have checked out the default version. Otherwise, it keeps the manifest exactly where it is. I think this is a worthwhile feature to emulate to fix this issue, thoughts?

We could add a warning that says `you have a non-default branch checked out, not updating the manifest` or so, with instructions on how to get back to self-updates if desired.

mbolivar, 2019-01-29:
@carlescufi @tejlmand is this still something you need at Nordic?

If not, can we please close? I still feel like the `git ls-remote` approach seems fine for upstream use, especially since that will include a `west.yml` with actual commits (or tags) for each external repository.

mbolivar, 2019-05-25:
@carlescufi @tejlmand given the lack of response here (and lack of questions about this since I've joined Nordic, hehe) I'm going to close this. Please reopen if this continues to be important.

closed 2019-05-25 by mbolivar