zephyrproject-rtos/west #46: west should set ZEPHYR_BASE, not look for it

mbolivar, 2018-09-23:
West knows where Zephyr is, since it parses the manifest.

This issue tracks extracting the manifest parsing code in the project command implementations into a helper module, where it can be used in the context validation code to find zephyr and set ZEPHYR_BASE in the environment before doing anything else.

carlescufi, 2018-09-24:
I assume this is the cause of the issue I'm seeing:

1. Install 0.2.0rc1 with pip
2. west init -u https://github.com/mbolivar/manifest --mr multirepo-test-manifest zephyrproject
3. source zephyr\zephyr-env.cmd
4. Build, flash successfully
5. cd %userprofile%
6. west --version:
```
fatal error: you must specify a command
usage: west [-h] [-z ZEPHYR_BASE] [-v] {flash,debug,debugserver} ...
```

mbolivar, 2018-09-24:
That's actually because the experimental west launchers in zephyr don't support the --version flag. This is being addressed in https://github.com/zephyrproject-rtos/zephyr/pull/10167.

tejlmand, 2018-09-25:
note, in case another manifest is used, which uses a forked version of zephyr it should be considered how west discovers which of the repo's in the manifest that is the correct zephyr.

Should it be required that:
```
path: zephyr
```
under the project containing Zephyr is used, if it is not Zephyr itself.

carlescufi, 2018-09-25:
As I see it we have the following issues to solve as part of this:

1. Make sure that users of older monorepo installations can continue to use `make/ninja flash`.
2. Make sure that users who do not want to use `west` directly can still build using CMake directly after the multirepo transition (provided they have cloned the correct repositories first).
3. Document that out-of-tree code (samples, drivers, etc) need to live within the current active wworkspace or ww (the folder F passed to `west init F`).

carlescufi, 2018-09-25:
@tejlmand 

Good point. Either fix the path to `zephyr` or add metadata to the manifest format.

mbolivar, 2018-09-26:
> Good point. Either fix the path to `zephyr` or add metadata to the manifest format.

Agree. We can have both though, if we want.

Proposal:

- the project named "zephyr" is ZEPHYR_BASE by default, if it exists.
- if your zephyr has another name, you need to add `is-zephyr-base: True` to the project definition in the manifest. At most one project shall have this flag set, or the manifest is invalid.
- a manifest without a ZEPHYR_BASE is invalid.

The diff to the manifest schema would be:

```
--- a/src/west/commands/schema.yml
+++ b/src/west/commands/schema.yml
@@ -133,3 +133,6 @@ mapping:
               clone-depth:
                 required: false
                 type: int
+              is-zephyr-base:
+                required: false
+                type: bool
```

An example manifest which used it would be:

```
manifest:
  remotes:
    - name: mbolivar
      url: https://github.com/mbolivar

  projects:
    - name: zephyr-for-upstreaming
      remote: mbolivar
      is-zephyr-base: True
      # Note "path: zephyr" is not required here -- west should be
      # able to deal with zephyr being wherever the manifest author
      # wants to put it.
```

Thoughts?

mbolivar, 2018-09-26:
> Thoughts?

If the build system ever gets smart enough to not need this, we can just start to ignore `is-zephyr-base`.

tejlmand, 2018-09-27:
> Proposal:
>  -   the project named "zephyr" is ZEPHYR_BASE by default, if it exists.
>  -  if your zephyr has another name, you need to add is-zephyr-base: True to the project definition in the [...]

Maybe have an additional assumption
- if a repository is checked out as `zephyr`, then it is assumed to be Zephyr base

That way one can also write:
```
manifest:
  remotes:
    - name: mbolivar
      url: https://github.com/mbolivar

  projects:
    - name: zephyr-for-upstreaming
      remote: mbolivar
      path: zephyr
```
in case zephyr is anyway clone into zephyr folder.

but being able to also specify with `is-zephyr-base: True` is good option, as it allows someone to use any name for zephyr the like.

tejlmand, 2018-09-27:
> - a manifest without a ZEPHYR_BASE is invalid.

Please no.
I could imagine that the `west` repository handling could be useful for other things then pure Zephyr, although it targets `zephyr`

Maybe a warning ?

mbolivar, 2018-09-27:
> Maybe have an additional assumption
> 
> * if a repository is checked out as `zephyr`, then it is assumed to be Zephyr base

Adding this rule to the list of assumptions would result in ambiguous results in some cases, for example:

```
  projects:
    - name: zephyr
      path: zephyr-path
    - name: something-unrelated
      path: zephyr
```

I picked "name" somewhat arbitrarily, but I think it's important that at most one of the "name" and "path" attributes determines the default ZEPHYR_BASE, to avoid ambiguities such as these.

> I could imagine that the west repository handling could be useful for other things then pure Zephyr, although it targets zephyr

I too ultimately want to use this tool for things other than Zephyr, but I am not sure it's a good idea to optimize for that potential future yet.

In practice, 100% of the users we care about should see a hard error if there is no repository to nominate as ZEPHYR_BASE. We can think about extracting the project components of west into their own program and competing with repo once we're wildly successful in a zephyr context ;).

tejlmand, 2018-09-28:
> > Maybe have an additional assumption
> > - if a repository is checked out as zephyr, then it is assumed to be Zephyr base
>
> Adding this rule to the list of assumptions would result in ambiguous results in some cases, for example:
> 
> `  projects:`
> `  - name: zephyr`
> `    path: zephyr-path`
> `   - name: something-unrelated`
> `     path: zephyr`
>
>  picked "name" somewhat arbitrarily, but I think it's important that at most one of the "name" and "path" attributes determines the default ZEPHYR_BASE, to avoid ambiguities such as these.

Agree, and in that case you can still place: ` is-zephyr-base: True` on the correct project, which should of course take precedence over anything else.

But of course, only supporting the ` is-zephyr-base: True` would avoid someone creating a manifest as in your example.

> >  I could imagine that the west repository handling could be useful for other things then pure Zephyr, although it targets zephyr
>
> I too ultimately want to use this tool for things other than Zephyr, but I am not sure it's a good idea to optimize for that potential future yet.

Yes, agree.
And i'm not against adding such a block if we think it helps the users.

I just don't like to add code, if we expect to remove it later anyway, unless it really helps the users.

mbolivar, 2018-09-28:
OK, so are we agreed on the proposal that a project named `zephyr` is ZEPHYR_BASE by default, unless explicitly set elsewhere?

carlescufi, 2018-09-28:
@mbolivar 

>OK, so are we agreed on the proposal that a project named zephyr is ZEPHYR_BASE by default, unless explicitly set elsewhere?

Yes, let's go with this assumption, it's the only reasonable one in my opinion.

mbolivar, 2018-10-02:
> > OK, so are we agreed on the proposal that a project named zephyr is ZEPHYR_BASE by default, unless explicitly set elsewhere?
> 
> Yes, let's go with this assumption, it's the only reasonable one in my opinion.

FTR, I went with "project with path zephyr" instead of "project named zephyr", as that seems to be closer to what people will expect. A project named "my-company-zephyr-fork" can still be cloned into the zephyr directory, and that should be ZEPHYR_BASE.

marc-hb, 2022-05-26:
Newer discussion on this topic in case a search engine lands you here:
- https://github.com/zephyrproject-rtos/zephyr/discussions/33521