zephyrproject-rtos/west #52: Should "west init" run "west fetch"?

mbolivar, 2018-09-25:
Right now, installing multi-repo Zephyr takes the following steps:

```
$ pip install west
$ west init <west-workspace>
$ cd <west-workspace>
$ west fetch
```

This issue tracks whether "west init" should run "west fetch" as well after cloning west and the manifest, to eliminate the last step before a complete installation is cloned.

I am ambivalent about this idea (generally speaking I don't like to add features to the bootstrapper) but thought it was recording the question (and its resolution along with rationale).

tejlmand, 2018-09-27:
I don't think so.

Let's keep them separate, as one might do a `west init`, but want certain arguments to the `fetch` command.
As example, manifest to use or revision of manifest.

carlescufi, 2018-09-27:
@myrdyr 
https://github.com/zephyrproject-rtos/west/issues/37

mbolivar, 2018-09-27:
@tejlmand @myrdyr thanks for the feedback! I've summarized your comments in the issue description.

I think at this point it's clear that it's not worth adding, so I'm going to close this with the answer "no". We can reopen if we have a compelling reason.

mbolivar, 2019-01-16:
Reopening with the new mental model. Should west init run west clone? (We moved away from west fetch because it created detached HEADS, which are gross).

tejlmand, 2019-01-28:
I think we should keep `west init` and `west update` seperate.

Reason: 
It allows users to do `west init` and then actually adjust `west.yml` to their needs if they wish so, before running `west update`

If users wants a one step approach, they can do:
`west init; west update`
or
`west init && west update`
depending on their platform.