zephyrproject-rtos/west #57: ImportError: No module named 'commands'

erwango, 2018-09-28:
When running west fetch, I get the following error:

<module>
    from commands import CommandContextError
ImportError: No module named 'commands'

Looking out on the web, i didn't found a way to install it, but ways not to use it:
https://github.com/yunabe/lgo/issues/42

And according to https://bbs.archlinux.org/viewtopic.php?id=125971, "module is deprecated and should be replaced by subprocess"

mbolivar, 2018-09-28:
@erwango thanks for the report! Actually 'commands' is an internal west module, we're just playing tricks with PYTHONPATH to import it without the 'west.commands' full qualifier. I think this is confusing and we should be importing all west module as west.xxx. Let me send a patch making that happen.

erwango, 2018-09-28:
Ok, so questin is how I ended up with this error then.
For info, here is what I got:
```
$ west fetch
Traceback (most recent call last):
  File "/local/home/frq07517/.local/bin/west", line 11, in <module>
    sys.exit(main())
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/bootstrap/main.py", line 206, in main
    wrap(wrap_argv)
  File "/local/home/frq07517/.local/lib/python3.5/site-packages/bootstrap/main.py", line 191, in wrap
    main_module = importlib.import_module('west.main')
  File "/usr/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 665, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/local/mcu/zephyr/tmp/zep3/west/src/west/main.py", line 19, in <module>
    from commands import CommandContextError
ImportError: No module named 'commands'
```

mbolivar, 2018-09-28:
Ah, I bet this is our README's fault. Did you 'pip install west==0.2.0rc2'? Right now we are in the prerelease period for 0.2.0 so the default release on pypi -- 0.1.0 -- may no longer be working.

erwango, 2018-09-28:
Indeed, I should be on the earlier one.
Let me clean this up and get back to you.

commit 166ed245 (2018-10-01): Import all west modules from west.

We are getting user feedback about missing dependencies (like
"commands" and "log") that are actually imports of internal west
modules by those names.

We've been playing games with the python path to omit typing "west",
but this always caused inconsistency (as the bootstrapper version is
imported as west._bootstrap.version) and now it's causing outright
user confusion.

Let's just be consistent and import everything as west.xxx.

Fixes #57.
closed 2018-10-01 by mbolivar