zephyrproject-rtos/west #59: West update usability issues

mbolivar, 2018-09-28:
The current auto-update behavior could use a bit of tweaking to make it more intuitive to those users who are editing their manifest or west repository and don't want to type --no-update all the time.

The issue is that it's not very convenient to switch to a manifest branch with custom changes and then issue 'west fetch', 'west pull', etc. commands, as that would try to update the manifest itself as well, which isn't what the user wants IMO. We could emulate the repo behavior instead, which is only to update the manifest itself to incorporate remote changes if the manifest repository is on the default branch tracking the remote which is created by west init.

This applies at least to the manifest, though for consistency it could be applied to west as well.

mbolivar, 2018-10-02:
@carlescufi @ulfalizer @tejlmand I actually lean towards calling this a bug -- I don't think it should block 0.2.0 but I do think we should fix it as soon as possible. Thoughts?

carlescufi, 2018-10-02:
>The issue is that it's not very convenient to switch to a manifest branch with custom changes and then issue 'west fetch', 'west pull', etc. commands, as that would try to update the manifest itself as well, which isn't what the user wants IMO

If you are switching to a manifest branch, you do want to pick up the new manifest from that branch, don't you? Or maybe I'm not following here

mbolivar, 2018-10-02:
> If you are switching to a manifest branch, you do want to pick up the new manifest from that branch, don't you? Or maybe I'm not following here

Here's the workflow I'm thinking of:

```
$ cd west/manifest
$ git checkout -b my-manifest-branch
$ $EDITOR default.yml   # add some projects, etc.; this is exactly the manifest I want now
$ west fetch
```

As a user, I expect `west fetch` not to rebase my manifest or otherwise touch it **in this case**, because it's checked out on a feature branch, not the original branch which tracks the remote I gave to `west init`.

This is how repo behaves, and I think it is the least surprising thing. If OTOH my manifest were on the 'default' branch which just tracks upstream, an update would make sense.

I think similar arguments apply to west itself.

carlescufi, 2018-10-03:
@mbolivar now I understand the issue. Absolutely. I concur with your assessment.

tejlmand, 2018-10-19:
How about the situation, where a user follows a given branch during development.
as example:
```
$ cd west/manifest
$ git checkout dev-branch-to-follow
$ west fetch
```
in case updates are made to the manifest on the remote branch, then you do want to fetch those changes.

I think getting back to the proposal here:
https://github.com/zephyrproject-rtos/west/issues/39#issuecomment-425189656
can help this.

Supporting, e.g.:
```
west fetch --mr <revision>
```
or in the case where `west init --mr <rev>` is given the revision, but then that requires a re-init in case you decide to switch manifest revision / branch using west.
(and I still think we should consider to allow revision change after `west init` as well as #39 to list tags / branches of the manifest without re-init)

If revision is given to `west fetch` as parameter, then `west` can know what to track.
And in case user does manual changes, then `west` should not update on `west fetch`

I think this issue starts to be related to discussion in #39

ulfalizer, 2018-10-19:
A start would be to skip the self-update if the `west.revision`/`manifest.revision` configuration values differ from the checked-out branch (and figure out all the corner-cases related to tags, SHAs, detached HEADs, etc., but I don't think it'll be that bad). I agree that that makes sense.

By the way, when thinking about West/manifest remotes and revisions and the like, I think it helps to focus on the configuration files first.

If we can get something that makes sense at the configuration file level, then the rest is just interfacing to them. If something doesn't make sense at the configuration file level, then that's a warning sign.

I.e., imagine that you're always just editing configuration files by hand. We might add something friendlier on top of that of course, but we should make sure that it makes sense at that level.

Maybe some per-command revision overrides and the like could make sense. Haven't really thought through it though.

ulfalizer, 2018-10-19:
Thinking about it some more, I think I've been using the wrong approach here all along. There's a much simpler thing we could do: https://github.com/zephyrproject-rtos/west/pull/81#discussion_r226635054

PR #82 (2018-10-24): commands: project: Only update west/manifest via fast-forward
If either the west or the manifest repository has local modifications,
skip automatically updating the modified repository, with a warning.
This avoids merge conflicts when working on them.

Maybe the manifest repository update could be skipped if the west
repository is out of date, and vice versa, but I suspect this version
might be good enough.

@ulfalizer please check if the issues below are addressed indeed.

Fixes #62 
Fixes #59
commit 4875ba36 (2018-10-24): commands: project: Only update west/manifest via fast-forward

If either the west or the manifest repository has local modifications,
skip automatically updating the modified repository, with a warning.
This avoids merge conflicts when working on them.

Maybe the manifest repository update could be skipped if the west
repository is out of date, and vice versa, but I suspect this version
might be good enough.

Fixes: #59
Fixes: #62
closed 2018-10-24 by carlescufi