zephyrproject-rtos/west #67: West needs a configuration file infrastructure

mbolivar, 2018-10-04:
We've discussed that west needs configuration.

Here is a proposed top-level docstring and feature description for the feature to get the conversation started:

```
'''West Configuration

West uses ini-style configuration files. A west configuration file is
named west.ini.

West searches for west.ini files in the following different locations,
in order:

1. Site-wide configuration directories (Linux only)

   The directories in $XDG_CONFIG_DIRS, if set, are checked for
   west.ini files in preference order (first listed has highest
   precedence, last listed has lowest).

2. Per-user configuration file(s).

   On Linux, this is $XDG_CONFIG_HOME/west/west.ini. $XDG_CONFIG_HOME
   defaults to $HOME/.config if empty or unset.

   On macOS, these are the files ~/Library/Application
   Support/west/west.ini and $HOME/.config/west/west.ini, in that
   order.

   On Windows, this is %LOCALAPPDATA%\west\west.ini.

3. Per-installation configuration file. This is the "west/west.ini"
   relative to the west installation root.

Configuration values in files that are found later take precedence
over values in earlier files. For example, per-installation
configuration files override any values in per-user configuration
files.
'''
```

**ini sections should correspond to each command**

So there would be a `[build]`,  `[flash]`, `[debug]`, etc section to configure the behavior of each. Examples:

```
[build]
# If not given on the command line, build chooses --board=nrf52_pca10040
board = nrf52_pca10040

[flash]
# If not given -- and if it is supported by the current build directory's CMake cache --
# the jlink runner would be used.
runner = jlink
```

**library behavior is also configurable**

E.g. maybe a `[runner.pyocd]` section to configure the behavior of that runner, like so:

```
[runner.pyocd]
# Override library default SWD frequency
frequency = 6000
```

**additional configuration mechanisms**

We should also provide a command-line switch and an environment variable to provide a configuration file.

E.g. `west --config=my-custom.ini <COMMAND> [options]`, or `WEST_CONFIG_FILES=my-custom-1.ini:my-custom-2.ini west <COMMAND> [options]`

ulfalizer, 2018-10-05:
Done any work on this?

mbolivar, 2018-10-05:
> Done any work on this?

Nothing substantial, go ahead if you'd like!

carlescufi, 2018-10-11:
@mbolivar I commented on the locations in order to match what Git uses here: https://github.com/zephyrproject-rtos/west/pull/75#issuecomment-428861903

PR #75 (2018-10-19): Get west and manifest revisions from a configuration file + support any revision format + misc.
```
Get west and manifest revisions from a configuration file

Add configuration file support via the standard Python configparser
module, which uses a Git-like .ini format. Use it to save the west and
manifest revisions specified when running 'west init', and use the saved
revisions when updating the west and manifest repositories.

Configuration file paths are based on Git, for consistency with the
Git-like commands. See the FILES section in git-config(1). Mac OS and
Windows paths were checked by  [... cut]
commit dfcbc4ad (2018-10-19): Get west and manifest revisions from a configuration file

Add configuration file support via the standard Python configparser
module, which uses a Git-like .ini format. Use it to save the west and
manifest revisions specified when running 'west init', and use the saved
revisions when updating the west and manifest repositories.

Configuration file paths are based on Git, for consistency with the
Git-like commands. See the FILES section in git-config(1). Mac OS and
Windows paths were checked by Carles Cufi with
'git config --list --show-origin'.

Add some new shorthands to _expand_shorthands() [... cut]
closed 2018-10-19 by mbolivar