zephyrproject-rtos/west #72: Detached HEAD when tracking a branch

carlescufi, 2018-10-08:
With a simple manifest that just tracks the master (or any other) branch of the repos included, I do not see why we end up with a detached HEAD upon running `west init` and `west fetch`. It would make sense to me to end up with a local branch checked out that matches the name of the remote tracking branch.

ulfalizer, 2018-10-08:
The thinking was that you'd do all your work on topic branches created with `west branch`, and that it might be confusing to create a local branch just to act as some kind of branch-after-init branch.

It's also nice to have all branches consistently track `manifest-rev`, as it means you can change the revision without things breaking, even changing it to an SHA if you want.

The only difference from `repo` is what we have the `manifest-rev` branch I think. Unless we're gonna work with a ton of SHAs everywhere, or switch between branches a lot in revisions, I like `repo`s design better (directly track upstream branches from work branches), but that's a different issue. :)

carlescufi, 2018-10-11:
Duplicate of #76

closed 2018-10-11 by carlescufi