zephyrproject-rtos/west #83: Possibly neater and simpler way of handling remotes

ulfalizer, 2018-10-22:
One thing I had somehow managed to miss all along is that Git allows URLs wherever names of remotes are allowed. You can do `git fetch https://github.com/zephyrproject-rtos/zephyr.git`, for example. That feature might even predate remotes, which are more like bookmarks (with some associated configuration settings, like where to store the refs that are fetched).

By always fetching directly from the URL, we could bypass the entire remote system. The nice thing about that is that we no longer rely on how remotes are configured. It's also really close to what @carlescufi wanted to see, only I thought it'd be a lot messier to implement due to remote management.

Another neat thing is that both of the following will just work (maybe they already do), with no extra work required:

```
[west]
remote = my-custom-remote
revision = master
```

```
[west]
remote = https://github.com/zephyrproject-rtos/west
revision = master
```

Any remotes we create would simply be for the user's convenience. They could even delete them or point them somewhere else, and things would still work.

ulfalizer, 2018-10-22:
This came up while looking at https://github.com/zephyrproject-rtos/west/issues/77 by the way. I'll see how well this works first, as it would help with that issue.

carlescufi, 2018-10-24:
That is a really good proposal and will solve a lot of issues that might arise from users that do not know how to manage Git remotes.
+1 to this.

carlescufi, 2018-10-24:
CC @mbolivar @tejlmand

tejlmand, 2018-10-25:
Agree, this looks very promising.
+1.

carlescufi, 2018-10-26:
> I also really like this idea. I don't understand what "bypass the entire remote system" means. Surely you aren't suggesting removal of that concept from the manifest schema?

No, I think @ulfalizer means not  using Git's indirection of `remote` -> `url`

mbolivar, 2018-10-26:
> No, I think @ulfalizer means not using Git's indirection of `remote` -> url

If that's the case that sounds great.

ulfalizer, 2018-10-26:
Yeah, it's just switching to specifying repositories via URLs instead of Git remotes. That removes remote management and gives less ways for things to break.

After the change, we never use Git remotes ourselves, except we create one for the user's sake.