zephyrproject-rtos/west #98: West crashes if default.yml doesn't exist

ulfalizer, 2018-11-06:
Looks like there's some new stuff in `set_zephyr_base()` that's causing it:

```
Traceback (most recent call last):
  File "/home/ulf/trywest/west/west/src/west/main.py", line 229, in <module>
    main()
  File "/home/ulf/trywest/west/west/src/west/main.py", line 201, in main
    args, unknown = parse_args(argv)
  File "/home/ulf/trywest/west/west/src/west/main.py", line 184, in parse_args
    set_zephyr_base(args)
  File "/home/ulf/trywest/west/west/src/west/main.py", line 90, in set_zephyr_base
    manifest = Manifest.from_file()
  File "/home/ulf/trywest/west/west/src/west/manifest.py", line 52, in from_file
    return Manifest(source_file=source_file)
  File "/home/ulf/trywest/west/west/src/west/manifest.py", line 81, in __init__
    with open(source_file, 'r') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/ulf/trywest/west/manifest/default.yml'
```

ulfalizer, 2018-11-06:
One bad thing about this is that you can't use `west init` to reset back to a working manifest, because it's using some new `west update` flags.

tejlmand, 2018-11-20:
@ulfalizer Proposed a fix in #99, which allows you to get out of the situation (but doesn't silence the stack trace)