zephyrproject-rtos/west #102: Projects in the west directory are not forbidden

mbolivar, 2018-11-08:
For example, even with forbidding projects named `west` or `manifest`, this manifest fragment is legal:

```
  projects:
    - name: not-west
      path: west
```

This creates a structure where the top-level `west` directory contains a `.git` for the `not-west` repository, which is very strange.

Arguably this is user error but the current behavior seems undesirable. We should consider what to do here:

- warn
- error
- skip the project?
- ...

mbolivar, 2018-11-20:
> And in future we could consider to add `--force` to `west` to allow user to overrule whatever restriction `west` poses

This would require adding `--force` to basically every invocation of west that touches a project or parses a manifest.

Perhaps a `manifest` configuration option?

tejlmand, 2018-11-20:
@mbolivar you're right, it would be required for every command, expect maybe `west list` ;)

I think my main point was to suggest the error for now, and then take it up, if someone needs this.

mbolivar, 2018-11-20:
Erroring out seems sane.

mbolivar, 2019-05-15:
Migrating to enhancement

marc-hb, 2024-10-11:
Since this issue was filed, WEST_DIR has been renamed to hidden directory `.west` by commit 045b591e17b1bcab and has changed A LOT in nature.

BUT I tested this again with `path: .west/foo` and west version 1.3 still did not mind. So this issue is still valid.

marc-hb, 2024-10-15:
Fix submitted by @pdgendt in #754

PR #754 (2024-10-21): manifest: Do not allow projects inside the west directory
It's not allowed to have projects inside the .west directory, add a check when parsing the manifest.

Fixes #102
closed 2024-10-21 by pdgendt