zephyrproject-rtos/west #105: west flash displays spurious ninja message

carlescufi, 2018-11-12:
When running `west flash` one sees a spurious `ninja: no work to do.` message

```
$ west flash
ninja: no work to do.
Using runner: nrfjprog
Flashing file: /home/carles/src/wncs/zephyr/samples/bluetooth/beacon/build/zephyr/zephyr.hex
Parsing hex file.
Erasing page at address 0x0.
Erasing page at address 0x1000.
Erasing page at address 0x2000.
Erasing page at address 0x3000.
Erasing page at address 0x4000.
Erasing page at address 0x5000.
Erasing page at address 0x6000.
Erasing page at address 0x7000.
Erasing page at address 0x8000.
Erasing page at address 0x9000.
Erasing page at address 0xA000.
Erasing page at address 0xB000.
Erasing page at address 0xC000.
Erasing page at address 0xD000.
Erasing page at address 0xE000.
Erasing page at address 0xF000.
Erasing page at address 0x10000.
Applying system reset.
Checking that the area to write is not protected.
Programming device.
Enabling pin reset.
Applying pin reset.
Board with serial number 682411272 flashed successfully.
```

mbolivar, 2018-11-14:
This is not a bug. West flash re-runs the build before flashing.

closed 2018-11-14 by mbolivar

carlescufi, 2018-11-14:
Thanks for the clarification.