zephyrproject-rtos/west #106: Determine west-based release strategy

mbolivar, 2018-11-14:
We mostly have a plan for tagging a west-based release as discussed in https://github.com/zephyrproject-rtos/west/issues/39#issuecomment-423629912.

The short story is:

1. Tag every project (zephyr gets v1.14.0, hypothetical-other-thing gets v1.14.0, etc.)
2. Set the revisions for each project in the manifest file to v1.14.0; tag the commit that does that v1.14.0 as well

Then you can do `west init --mr v1.14.0`.

I think the thing we did not realize is that this will not prevent users from continually getting updates to west itself! These may in turn be backwards incompatible with the v1.14.0 versions of things. So if the user ever runs 'west pull', say, that will update west itself, which is not safe.

This issue tracks resolving this; it's a blocker for Zephyr v1.14.

tejlmand, 2018-11-15:
Today a manifest may look as:
```
# The default West manifest file to be used for Zephyr.

manifest:
  defaults:
    remote: upstream
    revision: master

  remotes:
    - name: upstream
      url: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
```

how about extending this with `west` reference, such as
```
# The default West manifest file to be used for Zephyr.

manifest:
  <same as today>
  projects:
    - name: zephyr

west:
  revision: v.1.14.0
```
that would allow us to keep the knowledge in one place.
and introduce the rule, that if the west revision cannot be found, then west defaults to `master`.

This would allow to place the west revision in the manifest, tag it, and verify everything, as well as it won't break in case an invalid revision for `west` is placed in manifest, e.g. because there is work in progress. (Although a warning should be presented in that case.)

One could then place 
```
west:
  revision: v1.x.99
```
during development to indicate upcoming release.

tejlmand, 2018-11-20:
Thinking some more about my suggestion.
Placing the revision in the manifest will start to place more requirements on the bootstrapper, risking issues as seen in #98 and discovered during review in #99.

It is solvable, but we should be careful not to put too much logic into bootstrapper.

carlescufi, 2018-11-20:
@mbolivar @tejlmand 
The easiest way to solve this is to document that, in order to fetch a particular Zephyr version, one needs to also specify the west revision. This would also require to have the same tags on both west and zephyr. It's not very elegant but it would look like:

```
To check out Zephyr v1.14:
west init --mr v1.14.0 --wr v1.14.0
```

The other option, similar to this, is to introduce a new command-line option:
```
west init -r v1.14.0
```
This could do 2 things:

1. Check out both the manifest and west at v1.14.0
or
2. Check out the manifest at v1.14.0 and then use a mechanism (maybe in the manifest itself, maybe in a separate file) to find the corresponding west version.

mbolivar, 2018-11-20:
> It is solvable, but we should be careful not to put too much logic into bootstrapper.

@tejlmand yup, and for context, this is why we decided to have the bootstrapper special-case "know" where to find west itself, rather than putting the manifest parsing into the bootstrapper. I thought we were side-stepping a problem but it seems I was wrong and we've just moved it somewhere else.

mbolivar, 2018-11-21:
Thinking about it more, I have come around to your perspective.

We can maintain a separate schema for the west configuration, either keeping it in its own file (west.yml versus manifest.yml) or having a separate pykwalify schema for loading the value of the `manifest:` and `west:` keys. Neither seems to add unmanageable burden to the bootstrapper.

We will need to do a 0.4.0 for this though. Glad we haven't transitioned too hard yet...

(Pykwalify supports schema checking on either a YAML file or a loaded dict with Python data in it instead).

PR #125 (2018-11-28): Read west URL and revision from  manifest instead of config file
~Early RFC to get feedback, only partially tested~

Full implementation with additional proposal by @tejlmand:
https://github.com/carlescufi/west/pull/1

Also coupled to this PR to the manifest:
https://github.com/zephyrproject-rtos/manifest/pull/6

Fixes #106 (we will be able to simply tag the manifest with this PR)
closed 2018-11-28 by mbolivar