zephyrproject-rtos/west #112: Run a binary onto a board, d'oh

pfalcon, 2018-11-20:
I'm using whatever west is being run with "make flash" in today's Zephyr master. So, when I run:

> PYTHONPATH=/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/scripts/meta /usr/bin/python3 /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/scripts/meta/west/main.py flash --skip-rebuild --help

I get (among other output):

> Flash and run a binary onto a board.

I don't know what "run a binary onto a board" means. That makes me concerned that I'd better flash and run it myself, I only need west to tell me how (likely will be subject of the other ticket).

mbolivar, 2018-11-20:
Thanks for noticing the typo. I've sent a fix.

mbolivar, 2018-11-20:
Hi @pfalcon, I hope you'll forgive the piggybacking, but I've tacked another patch onto

mbolivar, 2018-11-20:
Sorry for the close/reopen, I pressed the "close and comment" button by mistake.

commit f3c10bb8 (2018-11-20): flash: fix a typo in the command description

Fixes: #112
closed 2018-11-20 by mbolivar