zephyrproject-rtos/west #113: Need a way for west to output flash, etc. commands for me

pfalcon, 2018-11-20:
Please see https://github.com/zephyrproject-rtos/west/issues/112 for not very relevant, but at least some context. The message conveyed there is "by various reasons, I don't trust various wrapper tools, and prefer to run my commands myself".

So, I came to this:

~~~
$ PYTHONPATH=/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/scripts/meta /usr/bin/python3 /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/scripts/meta/west/main.py -v flash --skip-rebuild
Using runner: jlink
Flashing Target Device
JLinkExe -if swd -speed auto -device MCIMXRT1052 -CommanderScript /tmp/tmp868see73jlink/runner.jlink
FATAL ERROR: 2 No such file or directory: 'JLinkExe'
Traceback (most recent call last):
  File "/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/scripts/meta/west/main.py", line 193, in <module>
    main()
...
~~~

So, the `-v` would seem to be just what I need, but of course after it quits, /tmp/tmp868see73jlink/runner.jlink doesn't any longer exist.

So, would need some kind of --dry-run option, which would just output commands to run, didn't run them, but created and left in place any dependent files. Thanks.

mbolivar, 2018-11-20:
> So, would need some kind of --dry-run option, which would just output commands to run, didn't run them, but created and left in place any dependent files. Thanks.

True `--dry-run` is not possible because subsequent commands sometimes depend on the output of previous ones (for some runners).

The intent is that `west -v flash` should be enough to see what really happened though, and I've sent a PR to fix this for the jlink case.

pfalcon, 2018-11-20:
And here's what I mean:

~~~
$ make BOARD=mimxrt1050_evk flash
[100%] Flashing mimxrt1050_evk
Using runner: jlink
Flashing Target Device
SEGGER J-Link Commander V6.40 (Compiled Oct 26 2018 15:08:38)
DLL version V6.40, compiled Oct 26 2018 15:08:28

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Cannot connect to J-Link via USB.

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Cannot connect to J-Link via USB.

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...FAILED: Cannot connect to J-Link via USB.
~~~

Please start with the assumption that all your "west flash" magic will fail, fail, and fail again. And the baseline service which west should provide is getting itself out of the way and letting user investigate directly what happens.

mbolivar, 2018-11-20:
> Please start with the assumption that all your "west flash" magic will fail, fail, and fail again. And the baseline service which west should provide is getting itself out of the way and letting user investigate directly what happens.

Yes, that is our goal, and always has been. I agree there is a problem with the jlink runner. Please check the PR at your convenience.

pfalcon, 2018-11-20:
> True --dry-run is not possible because subsequent commands sometimes depend on the output of previous ones (for some runners).

Well, step by step instructions like:

1. Run `JLinkExe ...`
2. Press green button in the top right corner of the board.

would work too ;-). Or output a 10-line shell script. Or 20-line. Just not 100 line.

> and I've sent a PR to fix this for the jlink case.

Thanks, will check it out. Apparently, when https://github.com/zephyrproject-rtos/zephyr/pull/11307 is merged.

mbolivar, 2018-11-20:
> Or output a 10-line shell script. Or 20-line. Just not 100 line.

The flash and debug commands used to run shell scripts in v1.9:

https://github.com/zephyrproject-rtos/zephyr/tree/v1.9.0/scripts/support

Some of them were over 100 lines, I'm afraid, so it seems there's no way to satisfy you on this one :stuck_out_tongue:.

They were rewritten in Python for v1.10 because Zephyr is a cross-platform project that works on Windows without any Unix wrappers:

https://github.com/zephyrproject-rtos/zephyr/tree/v1.10.0/scripts/support

This is the code base that eventually became west.

mbolivar, 2018-11-20:
> Thanks, will check it out. Apparently, when [zephyrproject-rtos/zephyr#11307](https://github.com/zephyrproject-rtos/zephyr/pull/11307) is merged.

Since the west PR has been approved and you've indicated you'll wait to review until #11307 is ready, I'm going to merge #115 now, which will close this issue. Please reopen if you feel appropriate.

commit e9938c5f (2018-11-20): runners: jlink: debug log commander script contents

We intend that 'west -v flash' provides sufficient information for the
motivaed user to stop using west if they so choose. The use of a
temporary file in the jlink runner whose contents are deleted as west
exits prevents that. Print its contents so we can let people get back
to work without west.

Fixes: #113
closed 2018-11-20 by mbolivar