zephyrproject-rtos/west #122: west checkout -b should respect my current HEAD

mbolivar, 2018-11-21:
If I manually make some commits in a random branch across multiple projcts, then run `west checkout -b NEWBRANCH`,  I expect NEWBRANCH to point at my current repository HEADs. However, it points at manifest-rev instead, making the behavior more like `git checkout -b NEWBRANCH manifest-rev` than `git checkout -b NEWBRANCH`.

This is confusing and annoying as now I have to go back and reset the new branches to where I wanted them to be in the first place.

mbolivar, 2018-11-21:
Note the new branch should still track manifest-rev -- but it should point at my current HEAD

ulfalizer, 2018-11-21:
Are we sure we want this? `west checkout -b` is our `repo start`, and is just a shorthand for `west branch` + `west checkout`. Branches created by `west branch` always start at the manifest revision as well.

When not working against the manifest revision, you can use normal Git commands to move stuff around.

I agree that the Git-like naming makes it a bit confusing in this case, but thinking functionality-wise.

mbolivar, 2018-11-21:
I've stumbled across this a few times now and been annoyed by the behavior each time.

mbolivar, 2018-11-21:
Repo start never worked the way I wanted either fwiw.