zephyrproject-rtos/west #124: Add build command to ease work with multiple boards

erwango, 2018-11-22:
I'm often building for multiple boards in same application and I like to keep the build for board_a when building for board_b.
Before west I was using a one arg custom command generating following tree:
```
hello_world / output / board_a / zephyr
                     / board_b / zephyr
```

Current way to do it with west is:
```
west build -b <board> -d <board>
```

Can we get a command to get the same while saving one arg (because I'm lazy)?

Could be:
```
west build -db <board>
```
or 
```
west build -B <board>
```
or
maybe `west build -b` could create `hello_world/<board>` path by default ?

mbolivar, 2018-12-18:
Hi @erwango! Sorry for the long silence, but I've been thinking about this.

------------

**TL;DR:**

What about adding a configuration file option to create the build directory from a format string instead of your proposal?

-------------

Build directory naming is something that can get kind of complicated even if it seems simple. In a separate tool I wrote that I think of as a west prototype, the default build directory for a Zephyr application is:

`build/{relative-path-to-source-directory}/{board}/app`

The final `/app` is because there's an `/mcuboot` directory next to it as well. This extra complication is worth it to me because I build multiple apps for multiple boards, along with separate MCUboot builds for each configuration, because I need application-specific DT overlays for each MCUboot build in general.

This default means I can easily get things like this hierarchy of build directories with a single build command in this other tool (not west!):

```
build
└── some-subdirectory
    ├── an-app
    │   ├── frdm_k64f
    │   │   ├── app
    │   │   └── mcuboot
    │   └── nrf52_pca10040
    │       ├── app
    │       └── mcuboot
    └── some-other-app
        ├── frdm_k64f
        │   ├── app
        │   └── mcuboot
        └── nrf52_pca10040
            ├── app
            └── mcuboot
```

Since I think that's where at least some users will end up eventually, I want to be careful about how we add smarter build directory generation to `west build` so it is future proof, while keeping the default set to `build` for simplicity and not confusing new west users who are comfortable with the usual Zephyr way of doing things.

Because of this, I think making it possible to specify a new build directory's path with a configuration file format string might work better.

You could then replace the default configuration value (which would stay `"build"`) to `"build/{board}"` by doing something like this:

```
$ cat >> west/config
[west.command.build]
build-dir-fmt = build/{board}
```

This option would only be used to create a new build directory with a default name, i.e. if:

1. -d is not given at the command line (highest precedence)
2. $PWD is not a zephyr build directory 
3. $PWD/build (the default build directory) is not a zephyr build directory (lowest precedence)
4. the directory implied by the `west.command.build.build-dir-fmt` option and command line arguments does not exist (i.e. if `west build --board=my-board` is given, `build-dir-fmt = build/{board}`, and `build/my-board` exists, use that)

Otherwise, the first of 1./2./3./4. which matched would 'win' instead. Breaking the tie if multiple build directories which match rule 4. exist would be done by build directory mtime, most recent first. (We could consider configurable tiebreaker policies as a later enhancement.)

I think eventually this would imply some changes in `west {flash,debug,debugserver,attach}` to make them aware of this as well, but I am not so sure the right way to handle that.

Anyway, if that proposal is okay with you, I'm thinking this would be a good set of initial format strings:

- `board`: the `--board` value
- `source-dir`: the relative path to the source directory (as given or inferred)

We could add more as we go on.

mbolivar, 2018-12-18:
Addendum: perhaps we could also add a `--bf / --build-dir-fmt` option to `west build` itself, which would override the config file option if given. If the option is given, it would unconditionally set the build directory. You could then say:

```
west build --board disco_l475_iot1 --bf 'build-{board}'
```

and it would do the right thing.

erwango, 2018-12-21:
@mbolivar 

I'm fine with your proposal. So I'd just need to do this configuration once, and it would apply to all my builds, whatever location I'm running west from ?

erwango, 2019-01-03:
> > I'm fine with your proposal. So I'd just need to do this configuration once, and it would apply to all my builds, whatever location I'm running west from ?
> 
> Correct.

Well this is a +1 for me then!

lemrey, 2019-02-06:
Yea, I second @erwango here; my build folder structure is identical to his: `prj/build/board`

I think that one important point here is which folder we assume west commands are run from.
Assuming that `west` commands are run from a project root sort of makes it more clunky to use.
I don't do anything from a project root, I run all `cmake` and `ninja` commands from a build folder and typically switch between build folders; I suppose @erwango does the same.

Basically, with this kind of hierarchy, from within a build folder it is possible to have commands with no arguments at all that will work for every board:
- `git nuke` the build directory
- `build` the project for the _right_ board, without any arguments (this is one a one line alias)
- `ninja flash/debug` the _right_ binary, without any arguments
- `cd -` switch build dir

@mbolivar your proposal would help a bit, but the user would still have to specify the board/build folder when invoking any `make`/`ninja` targets.

erwango, 2019-02-06:
@lemrey, actually I always call from prj root (where the CMakeLists.txt lives) with one argument which is the board (from which I derivate the build dir thanks to custom commands).

lemrey, 2019-02-06:
> @lemrey, actually I always call from prj root (where the CMakeLists.txt lives) with one argument which is the board (from which I derivate the build dir thanks to custom commands).

Right, then @mbolivar 's proposal would satisfy your use case fully I suppose; mine is a bit different and I have written more about it in #184. The main point is supporting a workflow from within a project's _build_ folder. To be clear, I am not arguing against the default behavior, just suggesting an enhancement that would suit people with a slightly different workflow.

closed 2019-06-10 by carlescufi