zephyrproject-rtos/west #126: Manifest Version field

tejlmand, 2018-11-27:
During work on PR#125 in order to settle on #106 it was identified that in order to be prepared in future for `west` bootstrapper and `west` itself to load older project, i.e. `west` v2.0.0 loading a manifest written when only `west` v1.0.0 was available.

Thus it is important that the manifest contains a manifest version field in order for future `west` versions to be able to load manifest based on old schemas where required fields might have changed.

tejlmand, 2018-11-29:
There are currently two parts of the manifest that might change:
1) The `west` section
     Changes to the section might require an update to the bootstrapped, and thus it must be possible for old bootstrapper to detect and suggest an update if section cannot be parsed, as well as a new bootstrapper should still be able to at least load an old manifest and fetch the `west` referenced in there.
2) manifest section.
    A manifest file could reference just HEAD of master as the `west` toll to use, so the `west` might need to be able to validate old sections.

Question now is, if there is only a single version field in manifest file, this could potentially result in upgrade of bootstrapper, whenever manifest section changes, which imho is undesired.

But having a version for each section might also proves undesired.
One solution could be to use major number, whenever bootstrapper must be updated, and minor number for changes only related to sections parsed by `west` tool (e.g. manifest section)

mbolivar, 2019-04-30:
Downgrading to low for now

PR #324 (2019-10-21): Manifest version check
A new attempt. GitHub won't let me reopen the previous pull request https://github.com/zephyrproject-rtos/west/pull/321, sorry.

Fixes: #126 

There are 3 different error paths on incompatible versions:

- help path: warn that extension command help will be missing and most commands won't run is printed
- error path: commands which are known to require the manifest (update, list, ...) print a fatal error with similar info for the help path
- watch-out path: commands which don't need the manifest (config, topdir) print a warning that they ought to work, but an upgrade is  [... cut]