zephyrproject-rtos/west #128: West ccreates and check out an ambigous branch

tejlmand, 2018-11-28:
If a manifest file points to tags as revision, then `west` will create a local branch with same naming, causing `git` to issue a warning.

Secondly, a tag is often used to identify a `release`, that is immutable, but creating a branch with same naming as the tag can give some users a wrong impression they might be working on the release.

Thoughts on how to handle this is welcome, e.g. 
- should it be identified when revision is a tag ? and then a tag is checked out as: `<tag>-local-work`

```
=== Creating repository for zephyr (zephyr/)
=== Fetching changes for zephyr (zephyr/)
...
Checking out files: 100% (11536/11536), done.
Note: checking out 'refs/heads/manifest-rev'.

=== Creating branch 'v1.13.99' in zephyr (zephyr/)
=== Checking out branch 'v1.13.99' in zephyr (zephyr/)
warning: refname 'v1.13.99' is ambiguous.
Switched to branch 'v1.13.99'
Your branch is up-to-date with 'manifest-rev'.
```

carlescufi, 2018-11-28:
I see no real solution to this other than explicitly coming up with a naming scheme that is "guaranteed" not to clash with an existing branch and at the same time makes it clear that this is a tag:

`_tag-<tag name>` perhaps?

tejlmand, 2018-11-29:
suggestion.

1) Try to detect if a tag is being checked out
2) If a tag is referenced, check it out as detached HEAD, as would be done normally by `git checkout <tag>`
3) Print a help message to user:
```
One or more revisions in project refers to tags and are now in Detached head
Create common branches across all repositories, e.g.:
west checkout -b <tag>-local-work
before start working
```

Or extended version, same as above, but give user the ability to directly accept:
```
Do you wish to create the branch '<tag>-local-work' across all repositories now [Y/n]:
```

tejlmand, 2019-01-28:
Closing this, as it is no longer valid with the latest design changes.

closed 2019-01-28 by tejlmand