zephyrproject-rtos/west #129: Detached HEAD in west and manifest if pointing to a tag

carlescufi, 2018-11-28:
The revisions to `west` and `manifest` and specified in the following manner:

* west: `west: revision:` element in the manifest file itself
* manifest: `--mr` passed to `west` in the command-line

If either of those points to a tag instead of a branch, we end up with a detached HEAD. This is not really a bug, but something we could consider improving.

carlescufi, 2018-11-28:
This is related to #128, except the bootstrapper doesn't force the creation of a branch like an installation does.

tejlmand, 2019-01-28:
is this still relevant, taking into account the latest design changes ?

hakonfam, 2019-02-25:
We need something similar to this in our team, that when we do "west update" we can `cd` to one of the configured repositories, and have git tracking the branch configured in west.yaml.

When working with large features in a distributed team, it is desired to maintain and share development branches in which all commits are kept (ie no force pushing).

This use case is not solved in todays west.