zephyrproject-rtos/west #135: west intercepts CTRL+C and kills gdb

ntavish, 2018-12-04:
This makes gdb not usable from the shell.

```
tavish@resurgam:~/repos/zephyr/samples/basic/blinky/build (blackmagic_probe_runner)$ west -v debug -r blackmagicprobe --gdb-serial=/dev/ttyACM0
ninja: no work to do.
Using runner: blackmagicprobe
WARNING: Runner blackmagicprobe is not configured for use with nrf52840_pca10056, this may not work
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gdb -ex 'set confirm off' -ex 'target extended-remote /dev/ttyACM0' -ex 'monitor swdp_scan' -ex 'attach 1' -ex 'file /home/tavish/repos/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf' -ex 'load /home/tavish/repos/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf' -ex 'break main' -ex run
GNU gdb (GDB) 7.11.0.20160511-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pokysdk-linux --target=arm-zephyr-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Remote debugging using /dev/ttyACM0
Target voltage: unknown
Available Targets:
No. Att Driver
 1      Nordic nRF52
 2      Nordic nRF52 Access Port
Attaching to Remote target
0x00001804 in ?? ()
Reading symbols from /home/tavish/repos/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf...done.
Loading section text, size 0xaa9a lma 0x0
Loading section .ARM.exidx, size 0x8 lma 0xaa9c
Loading section sw_isr_table, size 0x180 lma 0xaaa4
Loading section devconfig, size 0x84 lma 0xac24
Loading section rodata, size 0x5e8 lma 0xaca8
Loading section datas, size 0x228 lma 0xb290
Loading section initlevel, size 0x84 lma 0xb4b8
Loading section _k_sem_area, size 0x18 lma 0xb53c
Loading section _k_mutex_area, size 0x14 lma 0xb554
Loading section _k_queue_area, size 0x10 lma 0xb568
Loading section _net_buf_pool_area, size 0x40 lma 0xb578
Start address 0x197c, load size 46518
Transfer rate: 20 KB/sec, 830 bytes/write.
Breakpoint 1 at 0x0: main. (2 locations)
Starting program: /home/tavish/repos/zephyr/samples/basic/blinky/build/zephyr/zephyr.elf 
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, main () at ../src/main.c:18
18	{
(gdb) c
Continuing.
^CNOTE: you just ran a copy of west from /home/tavish/repos/zephyr/scripts; this will be removed from the Zephyr repository in the future. West is now developed separately.
Traceback (most recent call last):
  File "/home/tavish/repos/zephyr/scripts/west", line 115, in <module>
    main()
  File "/home/tavish/repos/zephyr/scripts/west", line 104, in main
    run_scripts_meta_west()
  File "/home/tavish/repos/zephyr/scripts/west", line 89, in run_scripts_meta_west
    'main.py')] + sys.argv[1:])
  File "/usr/lib/python3.6/subprocess.py", line 286, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 269, in call
    return p.wait(timeout=timeout)
  File "/usr/lib/python3.6/subprocess.py", line 1457, in wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.6/subprocess.py", line 1404, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt

Program received signal SIGINT, Interrupt.
k_cpu_idle () at /home/tavish/repos/zephyr/arch/arm/core/cpu_idle.S:135
135		bx lr
(gdb)
```

mbolivar, 2018-12-05:
Yup, just ran into this myself using the openocd runner.  I know it was working for the pyocd and jlink runners at least recently, so not sure if it's a runner specific issue or a general regression yet. 

I'm traveling this week but will try to look into this.

Are you for sure using the popen_ignore_int and/or run_server_and_client methods to spawn gdb processes appropriately in the black magic runner implementation?

mbolivar, 2018-12-18:
@ntavish this should be fixed in https://github.com/zephyrproject-rtos/west/pull/143 (and https://github.com/zephyrproject-rtos/zephyr/pull/12143 for the "old style" zephyr/scripts/west).

I've also taken the fix for the BMP runner, thanks for that!

Once #143 is merged, how about we update the copy of west in zephyr so the BMP runner is available to all users?

commit 124c64d3 (2018-12-18): bootstrap: fix 'west debug' regression

Commit f8bd521 ("Add self-update for West and the manifest") switched
from importing the west module and invoking main directly from the
same process as the bootstrapper to invoking it as a subprocess.
This breaks 'west debug'.

Runners which run GDB directly to debug a target need to be able to
ignore SIGINT while GDB is running, as Ctrl-C halts the target and
returns control to the (gdb) prompt instead of terminating the program
as usual.

Because of that, the runner code must be running in the same process
as the bootstrapper, or the parent (bootstrap [... cut]
closed 2018-12-18 by carlescufi