zephyrproject-rtos/west #144: Support useful environment variables in west forall

mbolivar, 2018-12-19:
repo forall defines a useful set of environment variables for the duration of each call:

- REPO_PROJECT is set to the unique name of the project.
- REPO_PATH is the path relative to the root of the client.
- REPO_REMOTE is the name of the remote system from the manifest.
- REPO_LREV is the name of the revision from the manifest, translated to a local tracking branch. Used if you need to pass the manifest revision to a locally executed git command.
- REPO_RREV is the name of the revision from the manifest, exactly as written in the manifest.

West forall should support equivalent features.

- WEST_PROJECT: project name (or path in the case of the manifest repository in the new multirepo scheme?)
- WEST_PATH: project path relative to the workdir root
- WEST_ABSPATH: absolute project path
- WEST_REMOTE_URL: the full URL of the remote as given in the manifest (since we use this internally instead of named remotes)
- Note that REPO_LREV is not necessary because we have a `manifest-rev` branch which can always be used
- WEST_MANIFEST_REVISION: equivalent to REPO_RREV
- WEST_MANIFEST_BRANCH: manifest tracking branch, if given

mkschreder, 2023-03-09:
It would also be nice to have something like "west forall-in-group" or something similar so that it is possible to run it only on a subset of projects.

mbolivar-nordic, 2023-03-10:
> It would also be nice to have something like "west forall-in-group" or something similar so that it is possible to run it only on a subset of projects.

Like this? https://github.com/zephyrproject-rtos/west/pull/641

mbolivar-nordic, 2023-03-10:
@mkschreder I'm going to wait to merge #641 until:

1. you confirm it's what you're looking for
2. after that, I have time to write some test cases

mkschreder, 2023-03-20:
> @mkschreder I'm going to wait to merge #641 until:
> 
> 1. you confirm it's what you're looking for
> 2. after that, I have time to write some test cases

Very nice. Yes I think group is probably the right way to approach this for now. I suppose we can view groups as tags of sorts where we can for example have a group called "public" where all repositories in that group would have a public trunk counterpart where latest local main trunk can be pushed using forall git push public origin/main:main. Or some other similar scenario.

marc-hb, 2024-08-28:
This was just merged:
- #144

closed 2024-08-30 by pdgendt