zephyrproject-rtos/west #149: Global west state makes it harder to unit test

mbolivar, 2019-01-10:
West has some global state that makes it harder to write proper unit tests. Examples:

- global configuration state is written into a module-level object in west.config
- log verbosity level is too

This makes it harder to write proper unit tests and causes other annoyances.

mbolivar, 2019-05-25:
This bugs me all the time when I write tests, and I'm constantly having to hack around it, as shown in #271. Bumping priority to medium as a result (since it's a problem that is slowing down development, despite not being user-visible).

mbolivar-nordic, 2023-02-27:
We're down to `west.log` at this point. We can close this issue once that deprecated module is removed sometime in the west 2.0 or later timeline.

marc-hb, 2024-10-10:
Progress from @pdgendt in https://github.com/zephyrproject-rtos/zephyr/pull/79171 and others.