zephyrproject-rtos/west #161: Error on first `west update` causes project to be in invalid state

tejlmand, 2019-01-28:
In case an error occurs during first `west update` on a sub-project, it causes this project to hang and subsequent calls to `west update` to fail.
This prevents the user from simply fix any issues in the manifest file and just re-run `west`

Work-around: manually remove the bad state repo folder and re-run `west update`.

Error seen:
```
=== Fetching changes for <repo> (<folder>/)
remote: Repository not found.
fatal: repository 'https://github.com/<repo>/' not found
FATAL ERROR: Command 'git fetch --tags -- https://github.com/<repo>' failed for <repo> (<folder>/)
```
after fixing the wrong remote for the project and re-running `west update`
the following error is seen (expected behavior, `west` would re-init the project)
```
fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
FATAL ERROR: Command 'git rev-parse --abbrev-ref HEAD' failed for <repo> (<folder>/)
```

mbolivar, 2019-01-29:
@tejlmand: in https://github.com/zephyrproject-rtos/west/pull/165, i propose a `--rebase` option whose implementation catches errors and accumulates them instead of bailing. Here is a snippet:

```
        failed_rebases = []

        for project in _projects(args, listed_must_be_cloned=False,
                                 exclude_manifest=True):
            # [...]
            if args.keep_descendants and is_ancestor:
                # [...]
            elif try_rebase:
                cp = _rebase(project, check=False)
                if cp.returncode:
                    failed_rebases.append(project)
                    _err(project, '{name_and_path} failed to rebase')
            else:
                # [...]

        if failed_rebases:
            # [...]
            raise CommandError(1)
```

I think we should use a similar strategy for the rest of the implementation, and the first step is to avoid calling `log.die()` anywhere except in a command handler, and throwing exceptions instead. We could just accumulate failed project updates in commands by catching exceptions or saving return codes the same way as failed rebases, letting as many projects as possible finish. The user would have to fix broken projects by hand.

Seem like the right approach?

tejlmand, 2019-02-25:
Fixed as part of v0.5.2

closed 2019-02-25 by tejlmand