zephyrproject-rtos/west #164: west update fails on Archlinux

ydamigos, 2019-01-28:
I run the following steps on Archlinux:

1. pip install --user west
2. west init zephyrproject
3. cd zephyrproject
4. west update

Step 4 fails with error
Traceback (most recent call last):
  File "${HOME}/projects/zephyrproject/west/west/src/west/main.py", line 23, in <module>
    from west import log
ImportError: cannot import name 'log' from 'west' (unknown location)

Does anyone have a similar issue?

ydamigos, 2019-01-28:
I had installed an older version of west (0.2.0rc2).
I uninstalled it (`pip uninstall west`) and run the steps below:
```
$ pip3 install --user west
Collecting west
  Downloading https://files.pythonhosted.org/packages/f9/5a/046a880614e62dfce305ce1a4ef8445d40f416eadd5f33cf14cf5f1a21db/west-0.5.0-py3-none-any.whl
Requirement already satisfied: colorama in /usr/lib/python3.7/site-packages (from west) (0.4.1)
Requirement already satisfied: pykwalify in /usr/lib/python3.7/site-packages (from west) (1.7.0)
Requirement already satisfied: PyYAML in /usr/lib/python3.7/site-packages (from west) (3.13)
Requirement already satisfied: docopt>=0.6.2 in /usr/lib/python3.7/site-packages (from pykwalify->west) (0.6.2)
Requirement already satisfied: python-dateutil>=2.4.2 in /usr/lib/python3.7/site-packages (from pykwalify->west) (2.7.5)
Requirement already satisfied: six>=1.5 in /usr/lib/python3.7/site-packages (from python-dateutil>=2.4.2->pykwalify->west) (1.12.0)
Installing collected packages: west
Successfully installed west-0.5.0
```

```
$ west init zephyrproject
Initializing in zephyrproject
=== Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr, rev. master into /tmp/manifest_0r3aeoen ===
Cloning into '/tmp/manifest_0r3aeoen'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 290288 (delta 7), reused 6 (delta 6), pack-reused 290272                                            
Receiving objects: 100% (290288/290288), 188.38 MiB | 1.42 MiB/s, done.                                           
Resolving deltas: 100% (212877/212877), done.
Checking out files: 100% (12766/12766), done.
Traceback (most recent call last):
  File "${HOME}/.local/bin/west", line 10, in <module>                                                   
    sys.exit(main())
  File "${HOME}/.local/lib/python3.7/site-packages/west/_bootstrap/main.py", line 407, in main           
    init(wrap_argv[1:])
  File "${HOME}/.local/lib/python3.7/site-packages/west/_bootstrap/main.py", line 186, in init           
    bootstrap(args)
  File "${HOME}/.local/lib/python3.7/site-packages/west/_bootstrap/main.py", line 248, in bootstrap      
    clone_west(os.path.join(tempdir, 'west.yml'), directory)                                                      
  File "${HOME}/.local/lib/python3.7/site-packages/west/_bootstrap/main.py", line 271, in clone_west     
    with open(manifest_file, 'r') as f:
```

Now, it fails during `west init`

carlescufi, 2019-01-28:
The current zephyr repository doesn't yet have a `west.yml`. Wait for the topic-west branch to be merged into master or run: `west init --mr topic-west`

ydamigos, 2019-01-29:
@carlescufi Thanks. Using `west init --mr topic-west`, I am able to clone a new copy of the zephyr repository using west.

closed 2019-01-29 by ydamigos