zephyrproject-rtos/west #168: when building with -DCMAKE_EXPORT_COMPILE_COMMANDS=ON, flash fails

nashif, 2019-01-29:
```
/home/nashif/.local/lib/python3.7/site-packages/cmake/data/bin/cmake -E env /home/nashif/.local/bin/west --verbose flash --skip-rebuild
WARNING: ZEPHYR_BASE=/home/nashif/Work/github/zephyr in the calling environment, but has been set to /home/nashif/Work/zephyrproject/zephyr instead by the manifest file /home/nashif/Work/zephyrproject/zephyr/west.yml
ZEPHYR_BASE=/home/nashif/Work/zephyrproject/zephyr (origin: manifest file /home/nashif/Work/zephyrproject/zephyr/west.yml)
Using runner: nrfjprog
FATAL ERROR: command flash cannot be run in this context: Runner nrfjprog received unknown arguments ['--verbose', 'flash']

```

mbolivar, 2019-01-29:
@nashif I believe that https://github.com/zephyrproject-rtos/west/pull/165 may fix your issue.

With that series, I can run:

```
west build --board frdm_k64f -s zephyr/samples/hello_world/ -- -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
west flash
```

With no issues. Assuming that's what you did, you should be fine when that is merged, as I can't run those commands successfully without pr 165.

mbolivar, 2019-01-29:
@nashif can you `west update` and retry? I just merged 165.

If that fails, can you paste the commands you ran as well, not just the error output?

mbolivar, 2019-01-29:
Closing as it was confirmed on IRC that this is a fix.

closed 2019-01-29 by mbolivar