zephyrproject-rtos/west #173: First 'west selfupdate' fails, second succeeds

mbolivar, 2019-01-29:
This also affects 'west update', of course.

Steps to reproduce:

```
$ cd $(west list -f="{abspath}" west)
$ git fetch upstream
$ git checkout upstream/master~3
$ west selfupdate
=== Updated west (.west/west/) to master (from https://github.com/zephyrproject-rtos/west).
/home/mbolivar/.local/share/virtualenvs/west/bin/python3: can't open file 'selfupdate': [Errno 2] No such file or directory
$ west selfupdate
```

The second one succeeds.

tejlmand, 2019-01-30:
Observation: when using `--reset-west` the selfupdate works on first try.
```
$ west selfupdate --reset-west
=== Fetching and resetting west (.west/west/) to 'master'
From https://github.com/zephyrproject-rtos/west
 * branch            master     -> FETCH_HEAD
```

closed 2019-01-30 by mbolivar

mbolivar, 2019-01-30:
@tejlmand should we close this, or keep it open until we get one more upstream west commit merged so we can test it starting from c5323934df286aaa2e3d400dba417b838bfa9d19?

tejlmand, 2019-01-30:
You already closed it :p 
Think it's fine to close, cause you can still test it, by checking out an older revision, and then apply this fix manually without committing, and then run `west selfupdate`

mbolivar, 2019-01-30:
Sorry, stupid combo buttons ;). OK, let's keep this closed, thanks.

PR #176 (2019-01-30): selfupdate: Fix for west selfupdate mechanism
Fixes: #173

This commit adds sys.argv[0] when restarting west after an update.
commit c5323934 (2019-01-30): selfupdate: Fix for west selfupdate mechanism

Fixes: #173

This commit adds sys.argv[0] when restarting west after an update.