zephyrproject-rtos/west #175: out-of-installation use of west

mbolivar, 2019-01-30:
User feedback noted that it's not possible by default to use west commands outside of the installation (i.e. not in a subdirectory of the directory containing .west).

We may be able to use the WEST_DIR environment variable to cover this case, so build, flash, etc. work normally even if things like the build directory are outside of the installation.

mbolivar, 2019-01-30:
Another potential solution is to allow the per-user or system-wide west configuration files specify where a default location to look for the west installation is, if one is not found where the command is run. Saves the need for yet-another-environment-variable since some classes of users have trouble with those.

mbolivar, 2019-01-30:
Relabeling this as a bug since it seems to be considered as such by a few users.

aurel32, 2019-01-30:
The main issue there is that the recommended directory layout to create an application (https://docs.zephyrproject.org/1.13.0/application/application.html) does not work anymore. In addition the error message is a bit cryptic in that case ("West version found in path does not support '/usr/bin/ninja flash', ensure west is installed and not only the bootstrapper. run 'west init' to fetch west.").

Determining WEST_DIR from the ZEPHYR_BASE would be nice to have, but if it is decided not to do so, we should at least:
- Improve the error message
- Update Zephyr's documentation about how to create an app
- Explain in the Zephyr's release the change and the new directory layout requirement.

mbolivar, 2019-01-30:
Hi @aurel32 and thanks for the comments! 

> The main issue there is that the recommended directory layout to create an application (https://docs.zephyrproject.org/1.13.0/application/application.html) does not work anymore. 

Right, but that is a zephyr issue since the documentation lives in the zephyr repository -- this issue is in the west repository so I am limiting scope to what affects west.

We are doing a lot of documentation work following up on the west transition and this will be fixed. I have even created a specific high priority issue to cover this exact problem now: The main issue there is that the recommended directory layout to create an application (https://docs.zephyrproject.org/1.13.0/application/application.html) does not work anymore. 

As you may know, we cannot make a release without closing all high priority bugs so there is a zero percent chance this will not get fixed :).

> In addition the error message is a bit cryptic in that case ("West version found in path does not support '/usr/bin/ninja flash', ensure west is installed and not only the bootstrapper. run 'west init' to fetch west.").
...
> Improve the error message

Great suggestion, thanks!

The rest are Zephyr problems so I think they should be handled outside of this repository's issue tracker.

carlescufi, 2019-01-30:
(from https://github.com/zephyrproject-rtos/zephyr/issues/12886)

>I don't like the idea of using ZEPHYR_BASE because that could point to the wrong tree if users use zephyr-env.sh.

I don't quite understand what the issue with using `ZEPHYR_BASE` to locate the west installation is. When building an out-of-tree application `ZEPHYR_BASE` is the link binding that isolated folder and the zephyr tree that is **required** to build it. So it only seems logical to me to expect that this `ZEPHYR_BASE` can also be used to retrace our steps back to west.

carlescufi, 2019-01-31:
CC @SebastianBoe @tejlmand

mbolivar, 2019-02-01:
@carlescufi  I've added another option in the issue description, which is to handle this in the user configuration file. I think that environment variables are confusing for Windows users. OTOH, everybody understands configuration files.

My hesitation about using ZEPHYR_BASE is that, in my experience when using multiple Zephyr trees on the same system, it is error-prone. When you `source zephyr-env.sh` or run `zephyr-env.cmd`, you get persistent state in your shell that can point to an unexpected place if you cd to the other tree.

This is one of the main reasons I decided to have west just set ZEPHYR_BASE properly to point at the zephyr within the installation for the duration of the call (and warn when that disagrees with the calling environment): it makes commands like `west build` and `west flash` work properly no matter what installation you invoke them from, without having to re-source an environment script. It also has the advantage that it can be expected to work when invoking west from another program when outside of any cmd.exe session, without setting system-wide environment variables.

Sourcing zephyr-env.sh is such an established practice at this point that it won't ever go away, which is why I don't like the idea of relying on a potential liar of an environment variable. That is why I think options 1 and 2 are best.

carlescufi, 2019-02-01:
>Another alternative if WEST_DIR is missing and west is invoked from outside an installation is to fall back on searching inside ZEPHYR_BASE, should that be defined.

That's the bit that is unclear to me. Regardless of west, ZEPHYR_BASE **has** to be defined and pointing to the correct zephyr tree, otherwise how is an out-of-tree application ever going to build? Today the whole premise of the location of Zephyr in the filesystem is based on ZEPHYR_BASE, so  unless we change that even if you go with option 1 the problem remains exactly the same.
The issue with the configuration file is that it might work for ninja flash and debug, but if you then start doing things like `cmake -G` everything will break apart, because that relies on ZEPHYR_BASE, won't it?

tejlmand, 2019-02-01:
A couple of thoughts on this.

Going forward, we wish to remove the need for the `west` repo, and distribute the repo-management part of `west` through PyPI, then `WEST_DIR` will not make sense for an ordinary user (we might consider `WEST_DIR` for those doing active `west` development)

When `west` is PyPI, then there will be no `.west/west` only a `.west/config` so I believe being able for users to specify their (main) Zephyr project using conf-file in `~/.westconfig` (Linux) makes very much sense, as long as this is an opt-in.

People playing with multiple Zephyr projects, might prefer to use environment variable `ZEPHYR_BASE`, and also I believe it is nice to be able to just re-direct in a single shell, without having to change the general setting.

Having the setting as: `~/.westconfig`
```
[zephyr]
zephyr_base = /path/to/zephyr
```
will help users building apps out of tree, and remove the need for typing `west --zephyr-base=/path/to/zephyr ....`

So my preference will be to support:
1. If inside a `west` managed project (ie. a parent folder contains `.west/config`), then Zephyr in there takes precedence over config setting
    - This allows a user to have a general project where `west` will default when OoT, but still being able to  go inside a local Zephyr project and run `west build`, e.g. ```cd <other/zephyr/>; west build```, without changing settings.
2. Support
```
[zephyr]
zephyr_base = /path/to/zephyr
```
3. Allow user to set `ZEPHYR_BASE` in the environment. And this way be using another `ZEPHYR_BASE` without changing conf setting.

As the conf-file is an opt-in, then user prefering `ZEPHYR_BASE` environment can do so.
Those just working on a single Zephyr project (or mainly a specific one), can set this in `~/.westconfig` if they want the ability to work OoT.

Everyone working inside the west handled project tree, don't need to do anything.

Additional:
I think the conffile should be backed by a (generic) conf command, e.g. `west config --user <key>=<value>`, in this example: `west config --user zephyr_base=</path/to/zephyr>`

carlescufi, 2019-02-04:
Conclusion after sync-up call, we agree that https://github.com/zephyrproject-rtos/west/issues/175#issuecomment-459687134 is the way forward:

The location of the zephyr repository when running any non-init west command out-of-tree (i.e. outside of a west installation with a `.west/`) is determined in this order:

- If `ZEPHYR_BASE` is set, that path is used
- If the option:
  ```
  [zephyr]
  zephyr_base = /path/to/zephyr
  ```
  is set, then use that path
- Else die with an error

mbolivar, 2019-02-04:
> * If the option:
>   ```
>   [zephyr]
>   zephyr_base = /path/to/zephyr
>   ```
>   is set, then use that path

On second thought, I think we need to drop this one. The problem is that the bootstrapper needs to find west, and it doesn't have access to the config module.

Once we have a unified multi-repo tool in what gets shipped via pypi, rather than just the bootstrapper, we can add that. For now I want to just support ZEPHYR_BASE as a fallback. That should be enough to support backwards compatibility with existing use cases.

mbolivar, 2019-02-04:
> If we do option 1. it would be great if it could not only do system-specific configuration for west, but also for CMake, so one could for instance put the zephyr-sdk intall directory in there.

@SebastianBoe I would dearly love to make west toolchain aware. That is on my radar for v1.15.

One could even imagine doing something like this so that the default toolchain can be configured on a per-board basis:

```
[toolchains]
gnuarmemb_toolchain_path = xyz
zephyr_sdk_install_dir = abc
cross_compile = qrs/my-target-triple-

[build]
board.<my_arm_board>.toolchain = gnuarmemb
board.<my_riscv32_board>.toolchain = cross-compile
board.<my_x86_board>.toolchain = zephyr
```

Behind the scenes, these would set ZEPHYR_TOOLCHAIN_VARIANT and the other corresponding variables at the CMake command line (`-DGNUARMEMB_TOOLCHAIN_PATH=xyz -DZEPHYR_SDK_INSTALL_DIR=abc -DCROSS_COMPILE=qrs/my-target-triple-`)

These defaults could be overridden at build time, e.g. to use the SDK instead of gnuarmemb for `my_arm_board`:

```
west build --board=my_arm_board -- -DZEPHYR_TOOLCHAIN_VARIANT=zephyr
```

Since user arguments are appended to the defaults chosen by `west build`, in this case, the toolchain variant would be `zephyr` instead of `gnuarmemb`.

Any early thoughts about the idea?

closed 2019-02-05 by carlescufi

SebastianBoe, 2019-02-07:
I like the idea of per-board toolchains, and a centralized system-configuration file.

And using "-D" would get the priorities right I believe.

But I would like it to be possible for this to be the recommended and standard way to do system configuration, but to achieve this we need it to be possible for users to invoke CMake and have CMake use the configuration file somehow.

tejlmand, 2019-02-07:
CMake is already capable of calling out to `west` if `west` is present, examples:
```
    86  if(WEST)
    87    # Verify that the west version found can be used for flashing, e.g. is not the bootstrapper.
    88    execute_process(COMMAND ${WEST} flash -h
    89                    RESULT_VARIABLE return_code
    90                    OUTPUT_QUIET
    91                    ERROR_QUIET
```
and in PR/13067
```
 execute_process(
    COMMAND
    west list --format={abspath}\;
    OUTPUT_VARIABLE
    west_project_list
  )
```
so if `west` exposes those settings through a call, eg. `west config --get <TOOLCHAIN>`
then CMake can use that info.
Environment varibles and command line arguments can take precedence over `west config --get`, so it stays consistent with what has been discussed earlier and agreed, see  https://github.com/zephyrproject-rtos/west/issues/175#issuecomment-459687134 and https://github.com/zephyrproject-rtos/west/issues/175#issuecomment-460299808

Also environment settings can be used if people don't want to use `west`

mbolivar, 2019-02-07:
>> But I would like it to be possible for this to be the recommended and standard way to do system configuration, but to achieve this we need it to be possible for users to invoke CMake and have CMake use the configuration file somehow.

Agree this is desirable.

> so if `west` exposes those settings through a call, eg. `west config --get <TOOLCHAIN>`
then CMake can use that info.

Agree this is a good mechanism. Making a `west config --get` would be easy, but `west config --set` is a heavier lift, since we need to round-trip whitespace and comments. That requires replacing configparser with a more sophisticated library we've found on pypi.

@SebastianBoe, @tejlmand, how do you want to proceed for roadmapping this? I can create west issues for

- west config
- toolchain awareness including per-board toolchains

Would that suffice for to add the build system tickets?

commit 80d5ab10 (2019-02-05): Fix use of west outside of an installation.

Fall back on ZEPHYR_BASE for backwards compatibility.  While we are
here, sync up the two versions of this function's way to get the final
"start" value, and improve the error message when we fail to find the
installation.

Fixes: #175