zephyrproject-rtos/west #180: source ~/.zephyrrc automatically when using west

alexanderwachter, 2019-02-04:
It would be useful if west checks for .zephyrrc file in the homedir and source it if present.

alexanderwachter, 2019-02-05:
my zephyrrc looks as following:
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
export GNUARMEMB_TOOLCHAIN_PATH=/usr/
because i'm not using the zephyr toolchain.

A configuration option that tells west what toolchain to use would be even better.
Maybe it could even ask for the toolchain when doing the west init command

tejlmand, 2019-02-25:
@alexanderwachter as discussed here:
https://github.com/zephyrproject-rtos/west/issues/198
https://github.com/zephyrproject-rtos/west/issues/175

The idea is to be able to specify this in west config files in future, and then add the posibility of letting  `CMake` fetch those settings.

This will further allow for global setting, while at same time allow for project local overwrites.
#215 is step one in this process.

mbolivar, 2019-10-11:
I'm going to close this for now since the ultimately proposed feature (west configuration options for the toolchain) is something we'd like to do for 'west build' in the zephyr repository.

closed 2019-10-11 by mbolivar