zephyrproject-rtos/west #184: automatically detect BOARD from pwd

lemrey, 2019-02-05:
I have been using a shell script for building Zephyr projects that automagically detects the `BOARD` from pwd and pass that to CMake. It has been working really nicely for me, especially because I find myself much more often with a terminal inside the build directory rather than the project's root (and I `git nuke`  often). I think it's more convenient than specifying the target board each time or making a bunch of aliases.

Example workflow:
`cd samples/blinky`
`mkdir -p build/nrf52_pca10040`
`cd build/nrf52_pca10040`
`build` (my shell script)
`ninja flash/debug`

Anytime I want to run the same project on a different board, I simply:

`mkdir ../nrf51_pca10028`
`cd ../nrf51_pca10028`
`build`
`ninja flash/debug`
`cd -`

repeat ad libitum.

I find it hard to reproduce this workflow with `west` and though I understand that is not a requirement at all, it could be a good enhancement:

`cd samples/blinky`
`west build --board nrf52_pca10040`
`west flash/debug`

The above is only three lines, that's nice! but attempting to switch a different board:

`west build --board nrf52840_pca10056`
`ERROR: Build directory [...] targets board nrf52840_pca10056, but board nrf52_pca10040 was specified`
(it could have created a new directory for the new board...)
`mkdir -p build/nrf52840_pca10056`
`west build --board nrf52840_pca10056 -d build/nrf52840_pca10056`
`west flash` <- which one is flashed? :) Let's try again
`west flash -d build/nrf52840_pca10056`

mbolivar, 2019-02-06:
> (it could have created a new directory for the new board...)

It doesn't know where, though. The default directory is 'build' and it should stay that way. This seems related to https://github.com/zephyrproject-rtos/west/issues/124.

> `west flash` <- which one is flashed? :) Let's try again

This is documented in the `west flash -h` output:

```
  -d BUILD_DIR, --build-dir BUILD_DIR
                        Build directory to obtain runner information from. If
                        not given, this command tries to use build/ and then
                        the current working directory, in that order.
```

I am not sure what you mean in the issue description by "detect BOARD from pwd".  If you are in a build directory, it will be used. So if you `cd build/nrf52840_pca10056; west flash` in your example above, west does detect BOARD from PWD.

Is something like the following what you are looking for?

```
cd samples/blinky
west build --board board1 -d build/1
west build --board board2 -d build/2
west flash -d build/1
west flash -d build/2
```

lemrey, 2019-02-06:
Hi @mbolivar, thanks for your reply. I'll try to explain more clearly what I mean.
Basically, `west flash` works quite well for me when it is run from inside a build directory.

What I think could be enhanced is `west` capability at building; `west build` is not able to determine which board to build for from the current directory path, specifically when the path is inside the build folder and named like the board.

`cd samples/blinky`
`mkdir build/nrf52840_pca10056`
`cd build/nrf52840_pca10056`
`west build`
`ERROR: source directory "." does not contain a CMakeLists.txt; [...]`
`FATAL ERROR: refusing to proceed without --force due to above error`
`west build -s ../..`
`ERROR: this looks like a new or clean build, please provide --board`
`FATAL ERROR: refusing to proceed without --force due to above error`

To get it do work, I eventually have to type:
`west build -s ../.. -board nrf52840_pca10056`

So within a build directory name like `/build/board`, `west build` requires a lot of information that it could determine itself.

From _outside_ a build folder and inside the project root folder, `west build` requires many arguments when maintaining multiple build folders; both board and build folder the first time, then I assume only the build folder afterwards (which also has to be specified everytime when calling `west flash`).

lemrey, 2019-02-06:
> It doesn't know where, though. The default directory is 'build' and it should stay that way. This seems related to #124.

Yes, they are related: make it easier to work with multiple build folders.
Which is what we all sort of do I suppose :)

carlescufi, 2019-06-05:
I honestly think this is going overboard with west's capabilities. I think this should be closed and implemented as a shell script instead, without west intervening at all. The features in https://github.com/zephyrproject-rtos/zephyr/pull/16636 should cover the basics in terms of configuring build folders, but when it comes to guessing boards I believe this opens too many potential pitfalls. @lemrey please close this if you agree.

closed 2019-06-06 by lemrey