zephyrproject-rtos/west #194: Should west-commands.yml be in the zephyr/ directory of each project?

mbolivar, 2019-02-08:
Right now, west extension commands are written into the manifest.

This makes maintaining downstreams involve a little bit more copy/pasting, since you need to find which repositories have extension commands and write them into your own manifest if you want them.

With the new 'modules' work that's going on, we're adding a special zephyr/ directory that the build system treats specially.

This issue is to decide whether west should look for a 'west-commands.yml' in that directory and use that to build its representation of extension commands.

My initial vote is "no" for the following reasons:

- that directory is for modules, and the zephyr repository has extension commands but is not a module
- every invocation of west would involve looping over every single project and doing file system lookups before anything else could happen, which would probably get old fast in installations with many projects

I thought it was worth asking here though for the record.

tejlmand, 2019-02-11:
@mbolivar  Great to raise this, so we have track of the discussion and decisions for future references.

> This makes maintaining downstreams involve a little bit more copy/pasting, since you need to find which repositories have extension commands and write them into your own manifest if you want them.

I don't think so.
Downstream anyway needs to look into `west.yml` in order to decide which repo to use downstream.
That might be, I need repo A and C, but not B.
Having the knowledge directly in `west.yml` that A contains extension commands, but not B and C, makes life easier when forking, imho.
Users don't need to look anywhere else, than the `west.yml` in order to be able to know if a project has extension commands.
 
> This issue is to decide whether west should look for a 'west-commands.yml' in that directory and use that to build its representation of extension commands.

I also vote _no_ .
Let's minmize the number of places where users needs to look when starting to fork.

>   that directory is for modules, and the zephyr repository has extension commands but is not a module

Completely agree on this.
Let's keep `west`-extensions far away from `zephyr/module.yml`.
I think it could be likely that someone could distribute additional repo, keeping track of, e.g. SDK tooling, and extend `west` with install commands for those tools.
Such a repo would not be a Zephyr module.

mbolivar, 2019-02-11:
:+1: sounds good @tejlmand. Since we are agreed I think I will close this. If we get this question again we can reopen this issue and continue discussion here.

closed 2019-02-11 by mbolivar