zephyrproject-rtos/west #197: unknown key "posixpath" in format string "{posixpath}"

SebastianBoe, 2019-02-11:
West is giving me this error:

```
Zephyr version: 1.13.99
-- Found PythonInterp: /home/sebo/bin/python3.4 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD nrf52840_pca10056
FATAL ERROR: unknown key "posixpath" in format string "{posixpath}"
-- Loading /home/sebo/ncs/zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts as base

```

Using latest Zephyr. Not sure which west version I am using or how to determine this.

> /home/sebo/ncs/zephyr/cmake/zephyr_module.cmake(15):  execute_process(COMMAND /home/sebo/.local/bin/west list --format={posixpath} OUTPUT_VARIABLE west_project_output )
> FATAL ERROR: unknown key "posixpath" in format string "{posixpath}"
>

tejlmand, 2019-02-11:
Related to this:
https://github.com/zephyrproject-rtos/zephyr/issues/13209

After updated Zephyr repo, users are now expected to run `west update` to ensure they are up -to-date with any sub-projects.
This is similar workflow to running a `git submodule update` on projects using sub-modules.

SebastianBoe, 2019-02-11:
Closing as duplicate

closed 2019-02-11 by SebastianBoe