zephyrproject-rtos/west #213: west init duplicates settings from ~/.westconfig

tejlmand, 2019-02-22:
Running `west init` will create the config file `.west/config` and populate this file with
```
[manifest]
path = <manifest-path>
```

However, if the file, `~/.westconfig` exists, then all settings from that file will also be copied into `.west/config` which basically means, that if user later changes the settings in `~/.westconfig` those settings will not be picked used, as they also exists inside `<project>/.west/config`

mbolivar, 2019-05-15:
@tejlmand I can't reproduce this on master:

```
$ cat ~/.westconfig
[zephyr]
base-prefer = configfile
[foo]
bar = baz

$ west init >/dev/null 2>&1
$ cat .west/config
[manifest]
path = zephyr
[zephyr]
base = zephyr
$ cat ~/.westconfig
[zephyr]
base-prefer = configfile
[foo]
bar = baz
```

Still valid for you?

mbolivar, 2019-05-25:
Closing since I can't reproduce.

closed 2019-05-25 by mbolivar