zephyrproject-rtos/west #217: Option for tracking branches instead of detached HEAD

hakonfam, 2019-02-25:
Add an option which allows users to specify that a given project should be checked out "directly". This means that instead of having DETACHED HEAD state in the git repository, the specified branch is checked out and set up to track the configured remote.

This is needed when working with large features in a distributed team.
In this situation, it is often desired to maintain and share development branches in which all commits are kept (ie no force pushing).

This should allow a developer to do the following:

```
cd west-root-project
git checkout feature-branch
west update
cd my-west-configured-repo
git status 
"Your branch is up to date with 'your_remote/feature-branch'."
# Do work
git add -u
git commit -sm "stuff"
git push
```

tejlmand, 2019-02-25:
This is intended to facilitate downstream work, where several developers may share a branch.

I suggest adding additional setting to the schema for the project map, so that one may specify:
```
  projects:
    - name: myproject
      revision: branchname
      checkout: true
```

and with an appropriate error message in case someone writes:

```
  projects:
    - name: myproject
      revision: SHA / tag
      checkout: true
```

Note: this is an opt-in for downstream projects, as branches which are forced pushed might experience undesired behavior if this option is set.

tejlmand, 2019-02-26:
Something else  to consider.

`west.yml` has changed (e.g. after a `git checkout` in manifest repo), and is now specifying a project where the branch shall be checked out by `west`

In case there already exists a branch with given name, what should `west` do?
Automatically assume that branch is to be controlled by `west`

This can be associated with some risk.

As example `master` is a common branch among all repos, and people switching between various remotes in manifetst repo, and doing a 
```
git fetch <remote/url> [refspec]
git checkout [refspec]
west update
```
could suddenly end in a situation where two different manifest specifies the same branch to checkout locally.

Instead of specifying this directly in `west.yml`, then users could opt-in for this behavior using `west config`, as example:
```
west config <project>.<branch>.<checkout> true
```
so one can specify:
```
west config net-tools.feature-branch.checkout true
```
Would then allow user to freely jump around with remotes and branches (including PRs) in manifest repo, and still obtain desired behavior for `<project>.<branch>` when that branch is present in `west.yml`

Note: allowing a two dot syntax will require an update to `west config` to allow a similar format as `git config where: `section.name.key` becomes:
```
[section "name"]
key
```

hakonfam, 2019-02-26:
I agree with this, this also allows the configuration to be left to the user. Which makes sense, as the role of the users differs, and not everyone would be interested in tracking the branch.

tejlmand, 2019-02-26:
some users might prefer to specify remotes where branches to be checked out, so that if `west.yml` points elsewhere, then `west` uses default behavior. 

so maybe consider the alternative:
```
west config <project>.<url-base>.checkout true
```
as well.