zephyrproject-rtos/west #221: Enhancement: merging west manifests

mbolivar, 2019-03-02:
This issue covers an enhancement to `west manifest` which would allow merging manifests. I think that it's useful to have on its own, and specifying and writing this command should be enough to define the right behavior for a multiple manifest west installation.

The portions of the signature specific to this enhancement are:

```
west manifest --merge [-o outfile.yml] MANIFEST1 MANIFEST2 [MANIFEST ...]
```

The command would "merge" the two or more input manifest files together, with an output file being a "combined" manifest. Output is to stdout by default, and to `outfile.yml` if given.

Specifically:

- All projects given in any of the input manifests would be present in the "merged" manifest output.
- A project is defined by its path in the installation, not its name or URL.
- Project-specific attributes (such as `clone-depth` and `path`) would take their values from the last manifest file in the argument list which set them. E.g. a `clone-depth` set in MANIFEST2 would override any value in MANIFEST1.
- The merged manifest file would take its default values from the *first* `defaults:` section which defines them. E.g. the default `remote` is given by the first input manifest which has a `defaults: remote: xxx` in the argument list. If a project in an individual manifest file takes a value from a default that is different from the merged manifest's default, the `west manifest --merge` file will just specify it exactly ("inline the default"). This is done as a heuristic to minimize the length of the generated manifest, because it's likely that upstream defaults will apply to the most projects once we split `zephyr/ext` into individual repositories. It doesn't affect the actual values that apply to each project.
- The merged manifest would take the values in the `west` section from the *last* manifest file which has one.

The above behavior does what I consider to be the "right thing" when doing something like this:

```
west manifest --merge zephyr/west.yml my-downstream/west.yml my-application/west.yml my-user-extensions/west.yml
```

I'm also pretty sure the operation defined this way is associative in terms of the final project attributes, but I still need to think through some details on how to merge two project lists, since the order of projects in the merged manifest matters for things like deciding which project "wins" if two different projects define an extension command with the same name.

TBD issue 1:

What if two different projects (as determined by the "path") have the same name? Options are:

1. error out,
2. generate duplicate names (since the name generates the URL, I don't think we can do anything else with the current manifest schema)

I think this is actually exposing a problem in the way I imitated Google repo when defining the west manifest schema. We may want to add another project definition syntax which allows each project to have a "name" which is unique in the installation, but specify its URL exactly instead of using `<remote url-base> + <name>`. Something like this:

```
remotes:
  - name: bar
    url-base: https://github.com/bar

projects:
  # Existing style, URL is https://github.com/bar/foo:
  - name: foo
    remote: bar
  # New alternative: specify URL exactly, like git submodules. It is
  # an error to also provide the "remote" key if "url" is given:
  - name: baz
    url: https://git.my-company.com/baz.git
```

TBD issue 2:

Are remotes first-class entities which can have their attributes overridden by later manifest files, or do they only apply to the projects in each of the individual input manifest files?

I think making them local to the inputs makes the most sense. The merged manifest can still have remotes defined, but that would just be a convenience to try to keep the length of the file shorter and make it easier to edit by hand further. The thing that "really" matters is the list of projects and their attributes.

Note that all this would imply some changes to the APIs in `west.manifest`, since a `Project` wouldn't necessarily have a remote (e.g. if its `url` is specified directly or if we decide to treat remotes as syntactic sugar to make typing a manifest shorter instead of first class entities in their own right). That's OK since we're not API stable though.

mbolivar, 2019-03-02:
Note also that this idea basically mandates introducing additional manifest file sections if we want to support features like Repo's `<default sync-j="4"/>` in west: we can't assume we can configure non-project west behavior in our `manifest: defaults: ...` anymore, as the `manifest: defaults:` section would be restricted to project attributes only.

tejlmand, 2019-03-04:
Thanks for raising this :)

A couple of comments for some clarification:
For project attributes, you state that last manifets wins.
> Project-specific attributes (such as clone-depth and path) would take their values from the last manifest file in the argument list which set them. E.g. a clone-depth set in MANIFEST2 would override any value in MANIFEST1.

but for the defaults, it is first manifest.
> The merged manifest file would take its default values from the first defaults: section which defines them. 

Why that difference ? And isn't there a risk associated with this, that users might find it hard to understand when a given setting is taken from first manifest or final manifest ?

Another question, a manifest may have
```
self:
    path: zephyr
```
but that is optional. So when combining multiple manifest files into a single, what will happen with `self` sections ?
Will they become independent projects ?
What will their `url` be ?
What if `self` is missing ?

tejlmand, 2019-03-04:
> Note also that this idea basically mandates introducing additional manifest file sections if we want to support features like Repo's <default sync-j="4"/>

Why does this need to go into the manifest ?
With #215 we could allow users the ability to specify: `west config --global sync-j 4` and then use the information in `west` when syncing projects

mbolivar, 2019-03-04:
Thanks for feedback!

> Why that difference ?

I tried to explain the rationale in the original issue description:

"This is done as a heuristic to minimize the length of the generated manifest, because it's likely that upstream defaults will apply to the most projects once we split zephyr/ext into individual repositories. It doesn't affect the actual values that apply to each project."

In other words, I expect that manifests which appear later in the list will have fewer projects, so my thought was to keep the length of the merged manifest shorter by taking defaults from the first project. That will make those defaults apply to the most projects possible.

E.g. I expect mainline to look something like this once we split ext/ up:

```
  projects:
    - name: net-tools
      revision: 30b7efa827b04d2e47840716b0372737fe7d6c92
    - name: ext1
      revision: abc
    - name: ext2
      revision: abc
    # ...
    - name: ext23
      revision: abc
    - name: mcuboot
      revision: abc
    - name: ci-tools
      revision: abc
    # ...
```

That's already more than 25 projects and can only increase as mainline zephyr gains more integrations, which can all be specified as a simple name/revision pair for now since they take a default remote.

(The number 23 in `ext23` is from how many sub-directories are inside `ext/*`, which is the appropriate level because it splits up per-vendor HALs and other projects from different sources into their own repositories.)

By contrast, the downstream repositories will probably have fewer than 10 projects.

So if the merged manifest takes defaults from the second, downstream manifest, all the `remote:` attributes in the 25 upstream projects have to be inlined. If it takes it from the first, upstream manifest, only 10 projects need 3 lines in the manifest to specify. So the resulting merged manifest is shorter if the first manifest's defaults win.

I'm not married to the idea though and if we decide there are good reasons that taking the last defaults is better even though the merged manifest will be longer, that's OK too. I think there are valid reasons on either side.

> And isn't there a risk associated with this, that users might find it hard to understand when a given setting is taken from first manifest or final manifest ?

I don't think so. The final manifest is self-contained and the defaults will be clearly specified at the top.

> Why does this need to go into the manifest ?
> With #215 we could allow users the ability to specify: west config --global sync-j 4 and then use the information in west when syncing projects

You're totally right. I forget repo doesn't have a separate config file system. This is definitely the way to go, +1 great idea.

tejlmand, 2019-03-06:
@mbolivar Thank you for the clarification and the example.

> The final manifest is self-contained and the defaults will be clearly specified at the top.

Which of course helps a lot as it is clear what the end result is.
I see the reason for keeping the final manifest as short as possible.

I assume this functionality is mostly targeted for automatic merging of file, e.g. for CI.

mbolivar, 2019-10-11:
I've begun work on this. It's going to be a release blocker for west 0.7.

tejlmand, 2019-12-19:
> whether we can load files from the file system when importing from projects, e.g. #330 (comment)

This is nice to have, far from need to have, so I suggest to keep out of `0.7`scope and maybe create a followup task so it can be addressed individually later :wink:

closed 2020-02-10 by mbolivar