zephyrproject-rtos/west #227: Provide command to check board supported features

erwango, 2019-03-12:
Would be nice to have a command to check supported features on one board.
Something like:
```$ west boardcheck nucleo_f334r8
  - arduino_i2c
  - gpio
  - i2c
  - spi
  - pwm
  - rtc
  - watchdog
  - counter
  - nvs
```

Output could be a dump of the board yaml or something more sophisticated.

mbolivar, 2019-03-12:
What would this look like? Is there a schema for board.yml? I didn't see one looking around. Do we just emit the `supported` key's value?

erwango, 2019-03-14:
@mbolivar ,

here is what `96b_carbon.yaml` looks like:
```yaml
identifier: 96b_carbon
name: 96Boards Carbon
type: mcu
arch: arm
toolchain:
  - zephyr
  - gnuarmemb
  - xtools
supported:
  - gpio
  - ble
  - i2c
  - rtc
  - counter
  - spi
  - usb_device
ram: 96
flash: 512
```