zephyrproject-rtos/west #231: `west` incorrectly sets `ZEPHYR_BASE` based on local config file

QKaseman, 2019-03-12:
As a result of https://github.com/zephyrproject-rtos/west/pull/211/, `west` incorrectly sets `ZEPHYR_BASE` based on the local config file in most cases (this is specifically caused by commit e17884f02f44b564919e5055b02f24ac32145137).

`zephyr.base` is set to the `path` in the manifest file for `zephyr` by `project.py` `PostInit` `do_run()`.

https://github.com/zephyrproject-rtos/west/blob/fdb634fd8f2bc017b9120d04e92876da305091e1/src/west/commands/project.py#L84-L86

 If you don't have `ZEPHYR_BASE` set and don't pass it in via the command line, `west` now checks the config file and tries to set `ZEPHYR_BASE` based on that config.

https://github.com/zephyrproject-rtos/west/blob/fdb634fd8f2bc017b9120d04e92876da305091e1/src/west/main.py#L416-L418

The value in the config (`zephyr` for the base `zephyr` project) is then passed to `os.path.abspath`, which gets the incorrect path for `ZEPHYR_BASE` depending on where you run it. 

https://github.com/zephyrproject-rtos/west/blob/fdb634fd8f2bc017b9120d04e92876da305091e1/src/west/main.py#L441-L442

For example, running it in `zephyr/samples/hello_world` results in `ZEPHYR_BASE` being set to `zephyr/samples/hello_world/zephyr`:

```
kasemq@kaer-morhen ‹ master › : ~work/z2/zephyr/samples/hello_world
[0] % west build -b nrf52840_pca10056 -s .
source directory: /home/kasemq/work/z2/zephyr/samples/hello_world
build directory: /home/kasemq/work/z2/zephyr/samples/hello_world/build (created)
BOARD: nrf52840_pca10056
CMake Error at CMakeLists.txt:3 (include):
  include could not find load file:

    /home/kasemq/work/z2/zephyr/samples/hello_world/zephyr/cmake/app/boilerplate.cmake
```

This can be reproduced with the following steps:

```
mkdir z2
cd z2
west init
west update
cd zephyr/samples/hello_world
west build -b nrf52840_pca10056 -s .
```

It does work correctly if you run the build from the `west` top directory (that contains `.west`).

Not sure if the correct fix is to store the absolute path in the `west` config file or assume the `west` config file path is a relative from `west_topdir` within `set_zephyr_base()`

mbolivar, 2019-03-13:
@carlescufi @tejlmand I think we need to address this for v1.14

tejlmand, 2019-03-14:
Fixed here: https://github.com/zephyrproject-rtos/west/pull/232

tejlmand, 2019-03-14:
> Not sure if the correct fix is to store the absolute path in the west config file or assume the west config file path is a relative from west_topdir within set_zephyr_base()

the problem with using an absolute path in `.west/config` is that it will fail if user later does:
`mv <multi-repo-project> /somewhere/else/<multi-repo-project>`
whould suddenly mean an absolute path points to wrong location.

PR #232 (2019-03-14): zephyr base: fixed west config zephyr.base when relative
This commit ensures that zephyr.base will be joined with west_topdir()
when a relative path is used in west config.

The previous solution was using os.path.abspath which would give
different results depending on current location when invoking west
command.

The advantage of allowing relative path in .west/config over requiring
an absolute path, is that it allows a user to rename a project's parent
folders without having to update zephyr.base.

Fixes #231
commit aa8c1836 (2019-03-14): zephyr base: fixed west config zephyr.base when relative

Fixes: #231

This commit ensures that zephyr.base will be joined with west_topdir()
when a relative path is used in west config.

The previous solution was using os.path.abspath which would give
different results depending on current location when invoking west
command.

The advantage of allowing relative path in .west/config over requiring
an absolute path, is that it allows a user to rename a project's parent
folders without having to update zephyr.base.
closed 2019-03-14 by carlescufi