zephyrproject-rtos/west #233: `west info` command

tejlmand, 2019-03-14:
Having a `west info` command available may be useful in order to expose several informations about the `west` installation an the project installled.

Those info can also be used by build build systems or CI,

Such information that can be useful:
- west topdir
- board listing

Feel free to add suggestions in comments

mbolivar, 2019-03-14:
At first thought, I would prefer to have separate commands for the items listed in the issue description. We can hide them by default (porcelain and plumbing) if they're too noisy for the `west -h` output.

tejlmand, 2019-03-14:
not sure on separate commands.
Cause the more commands of such style we create, the higher the risk that at some point, a new _info_ command is added that collides with a potential downstream extension command.

Which make me actually consider, should a `west info` output be allowed to fetch info from extension command ? 
so that if a project contains a `west info` as extension, then the output from that command can be concatenated with built-in `west info`

mbolivar, 2019-05-25:
This needs to be teased apart into what's core west versus what's an extension command. Removing good first issue label because of that.

mbolivar, 2019-10-11:
Closing. `west topdir` is now implemented, and a separate zephyr issue for board info is https://github.com/zephyrproject-rtos/zephyr/issues/19760

closed 2019-10-11 by mbolivar