zephyrproject-rtos/west #237: projects root should be configurable

galak, 2019-03-27:
Right now the assumption is that the root of projects is relative to the .west dir.  There should be a config option one can set in .west/ that overrides the default behavior of where the projects root is assumed to be.

galak, 2019-03-27:
Something like:

in `.west/config`:

```
[projects]
root = /path/to/projects
```

marc-hb, 2019-03-27:
Here's how it looked last time I used west:
```
$ tree -L 1 -a west-top/ # actually: manually filtered -L 2

west-top/
├── net-tools
│   ├── .git
|   ...
├── .west
│   ├── config
│   └── west
└── zephyr
    ├── .git
```

So I guess the request is about moving the entire `.west/` *outside* the current west top level? I mean not in some other place still inside west top. Some ASCII art like this in the description wouldn't hurt.

"Projects" is a confusing name not just because it's generally vague but more specifically because it's not clear whether the multiple "projects" refer to: 1. multiple git repos in the same manifest; versus:  2. multiple manifests and west tops for unrelated products/projects. The documentation - ping @dbkinder - mixes up both 1. and 2. in the same section: https://docs.zephyrproject.org/latest/tools/west/repo-tool.html 

> projects, which are Git repositories specified by west.yml

 yet the west top is called `zephyrproject`: singular.

"west top" is nowhere near that confusing but maybe there's an even better name.

mbolivar, 2019-03-28:
> The documentation - ping @dbkinder - mixes up both 1. and 2. in the same section: https://docs.zephyrproject.org/latest/tools/west/repo-tool.html

This is our fault, not David's. I've been trying to make time to write proper west documentation for a while and finally got to it: https://github.com/zephyrproject-rtos/zephyr/pull/14983

> "west top" is nowhere near that confusing but maybe there's an even better name.

We (the west developers) have been using the term "west installation" for short and "west installation root directory" if you're not into the whole brevity thing. In my opinion, the term "west installation" hasn't been very clearly documented, so I am not surprised there's confusion. I've tried to fix that in the above PR; feedback welcome.

So I would say actually that this is not about moving `.west` somewhere inside a project. I am strongly against that idea. A west installation is not a git repository with submodules -- it's more like a repo checkout -- and the zephyr repository is not the installation's root directory.

However, it might make sense to add a configuration option that lets you configure where to clone projects inside of the installation, rather than just assuming the installation root directory.

So, making up names here, if you did `west config --local project.base-dir zephyr`, then the `path` element in each of the `projects` in the manifest would be relative to `$INSTALLATION_ROOT/zephyr` rather than the current behavior, which is relative to `$INSTALLATION_ROOT`.

@galak is that what you are asking for?

galak, 2019-03-28:
> So, making up names here, if you did `west config --local project.base-dir zephyr`, then the `path` element in each of the `projects` in the manifest would be relative to `$INSTALLATION_ROOT/zephyr` [...]

Yes, I assume it can either be a relative or absolute path.  So I can also do:

`west config --local project.base-dir /home/kumar/my_module_dir`

mbolivar, 2019-03-28:
> Yes, I assume it can either be a relative or absolute path. So I can also do:

It's not really in line with the repo-style model, but we can try to support absolute paths as well. I wouldn't be shocked if there are weird edge cases, though :smile:

marc-hb, 2019-05-31:
> west config --local project.base-dir /home/kumar/my_module_dir

I'm even more confused. The first sentence at the top refers to the "root of (all) projects", now this seems to be referring to only one of them.
Again some simple ASCII art could save a lot of comments.

carlescufi, 2019-06-05:
@marc-hb it would refer to all projects that are not the manifest project. So I believe what @galak wants it to have:

```
/home/galak/src/zephyr/
├── .west
│   ├── config
│   └── west
└── zephyr
    ├── .git

/home/galak/modules/
├── net-tools
│   ├── .git
├── tinycbor
│   ├── .git
|   ...
```

marc-hb, 2019-06-05:
If you relax the constraint that modules can't be higher than .west/, then you might as well go all the way and also relax the constraint that modules should be next to each other. The complexity increase feels  similar and the vast number of additional west bugs should be in the same order of magnitude.

mbolivar, 2019-12-19:
I don't think this issue should be implemented. This is not how west is designed.

closed 2019-12-19 by mbolivar