zephyrproject-rtos/west #244: Command to import external module

microbuilder, 2019-04-01:
One of the more useful features of `west` is the ability to manage multiple 3rd party Zephyr modules via git repositories, and request specific versions of the repo.

At present, this requires adding a managed module dependency requires the addiition of the the repository to the **`manifest/remotes`** group, and the repo name and version to the **`manifest/projects`** group in `zephyr/west.yml`, as [shown here](https://github.com/zscilib/zscilib#adding-zscilib-to-your-project-via-west).

A command to automatically clone a repo from a URL, and merge the repo dependency into `zephyr/west.yml` might make this less error-prone manually editing the .yml file, and highlight the possibility of managing repos outside the core zephyr repo?

Also, the process of adding dependencies (see link above) isn't clearly documented at present, and it might be worth considering and example of how to do this in the official documentation.

mbolivar, 2019-05-15:
Hi @microbuilder and thanks for the report! Sorry for the long delay in responding.

A few things:

> At present, adding a managed module dependency requires the addition of the repository account to the manifest/remotes group,

In https://github.com/zephyrproject-rtos/west/pull/266, we have a commit which allows you to add projects by URL, like this:

```
manifest:
  projects:
    - name: my-cool-project
      url: https://github.com/my-github-username/my-cool-project-fork
```

> and the repo name and version to the manifest/projects group in zephyr/west.yml, as shown here.

Just FYI, only the name (and, as of #266, either the remote or url) is required. The version defaults to `master`.

> A command to automatically clone a repo from a URL, and merge the repo dependency into zephyr/west.yml might make this less error-prone manually editing the .yml file, and highlight the possibility of managing repos outside the core zephyr repo?

This is the main request in this issue, as far as I can tell.

However, it would be a significant departure from our principle to never edit the contents of the manifest repository or the manifest file itself. This is in contrast to commands like `repo sync` which will do things to the manifest repo that seem well specified, but often aren't in subtle ways that can make you crazy.

So I see the fact that we don't change the manifest today as a big feature, and I don't personally believe that the convenience of a command (which is not much IMO, given that we have good docs for the manifest format and multirepo pieces of west) is worth losing that. Do you feel strongly that this command is worth having?

> Also, the process of adding dependencies (see link above) isn't clearly documented at present, and it might be worth considering and example of how to do this in the official documentation.

We added a lot of documentation in time for v1.14, in particular:

https://docs.zephyrproject.org/latest/guides/west/repo-tool.html
https://docs.zephyrproject.org/latest/guides/west/manifest.html

I hope that you're OK with the docs now. If not, please LMK what's unclear and I'll try to fix it.

mbolivar, 2019-05-15:
> Do you feel strongly that this command is worth having?

If not, please close this issue and thanks again!

microbuilder, 2019-05-15:
@mbolivar The predates the recent documentation, and it's definitely a lot clearer now, thanks.

closed 2019-05-15 by microbuilder