zephyrproject-rtos/west #246: Support use cases other than Zephyr RTOS

mbolivar, 2019-04-02:
West can be already used for other purposes than Zephyr development, even without sacrificing the parts of it that make it convenient for Zephyr.

This issue tracks first-class support for that. The main issue I'm aware of is this error:

```
ERROR: no --zephyr-base given, ZEPHYR_BASE is unset, west config contains no zephyr.base setting, and no manifest project has path "zephyr"
```

It would be nice to generalize the existing mechanism for setting environment variables based on the contents of the manifest so you could do something like this:

```
manifest:
  self:
    path: zephyr
    config-vars:
      - zephyr.base: {abspath}
```

Where the `config-vars` key in the manifest's `self` subsection (or any `project` element) would be a list of `config_variable: value` pairs, where each `value` is a format string like that passable to `west list --format`. These values would be set at `west update` time, as if with `west config --local config_variable $(west list --format="value" path_to_project)`, provided that they do not already have values.

bengartner, 2019-05-02:
I'm trying to use west to manage a closed source project and I'm noticing a few things when I try to use my own manifest file. This may belong in its own issue as it doesn't seem related to the ZEPHYR_BASE issue itself.

First, the init script complains about the command "post-init." I can't tell why it's trying to run this command, if it's a custom command defined in zephyr somewhere, or what I don't get this error when using a zephyr manifest file.

```
From https://github.com/zephyrproject-rtos/west                                                                                                                           
* branch            master     -> FETCH_HEAD                                                                                                                             
* [new branch]      master     -> origin/master                                                                                                                          
* [new tag]         v0.1.0     -> v0.1.0                                                                                                                                 
* [new tag]         v0.2.0     -> v0.2.0                                                                                                                                 
* [new tag]         v0.2.0rc1  -> v0.2.0rc1                                                                                                                              
* [new tag]         v0.2.0rc2  -> v0.2.0rc2                                                                                                                              
* [new tag]         v0.2.0rc3  -> v0.2.0rc3                                                                                                                              
* [new tag]         v0.3.0     -> v0.3.0                                                                                                                                 
* [new tag]         v0.3.0rc1  -> v0.3.0rc1                                                                                                                              
* [new tag]         v0.4.0     -> v0.4.0                                                                                                                                 
* [new tag]         v0.4.0rc1  -> v0.4.0rc1                                                                                                                              
* [new tag]         v0.5.0     -> v0.5.0                                                                                                                                 
* [new tag]         v0.5.1     -> v0.5.1                                                                                                                                 
* [new tag]         v0.5.2     -> v0.5.2                                                                                                                                 
* [new tag]         v0.5.3     -> v0.5.3                                                                                                                                 
* [new tag]         v0.5.4     -> v0.5.4                                                                                                                                 
* [new tag]         v0.5.5     -> v0.5.5                                                                                                                                 
* [new tag]         v0.5.6     -> v0.5.6                                                                                                                                 
* [new tag]         v0.5.7     -> v0.5.7                                                                                                                                 
* [new tag]         v0.5.8     -> v0.5.8                                                                                                                                
cb660b6e28ce96c956e19a1244dac1df938e5ba3 refs/remotes/origin/master                                                                                                      
Already on 'master'                                                                                                                                                      
Branch 'master' set up to track remote branch 'master' from 'origin'.                                                                                                    
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V]  ...                                                                                                                
west: error: argument : invalid choice: 'post-init' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 
'forall', 'config', 'selfupdate', 'help')
--
```

I'm assuming this leaves the west installation in a bad state, because after this it complains about west config settings:
```
$ west update                                                                                                                                                            
FATAL ERROR: Parsing of manifest file failed during command update : missing key: 'manifest' in west config file
```
Which makes sense, as I haven't defined a .westconfig file, but the zephyr manifest seem to at least create a default .west/config correctly, and I also can't tell where, if anywhere, in the repo that's coming from.

Any suggestions on how to resolve this issues would be welcome.

bengartner, 2019-05-03:
I updated to the master instead of the 0.5.8, and it seems like both these issues are resolved. Possibly as a side-effect of removing bootstrapping. So probably no reason to clean track these unless someone sees them in 0.6.0+.

mbolivar, 2019-05-25:
I've been thinking about this a bit more.

- I think `set_zephyr_base()` should be moved to the zephyr repository, with all of its extensions inheriting from a class that calls it before their current `do_run()` code is invoked
- Unfortunately, I think it's too late for west to stop using ZEPHYR_BASE in the environment as a place to find the west installation, as that would break existing users that rely on environment variables and still want to use `ninja flash` for zephyr application directories that aren't inside the west installation containing their zephyr repository. But that's OK, and can be documented as a historical artifact. People who use west without zephyr aren't ever going to be likely to notice.

Since Zephyr's build system enforces a minimum west revision when it finds that the tool is installed, I think we can get away with this by moving `set_zephyr_base()` over to zephyr, deleting it from west and cutting a release, then bumping zephyr's minimum west.

bengartner, 2019-05-26:
@mbolivar I *think* there's a bit of a bootstrapping problem for non-Zephyr projects when you don't already have a west config file defined somewhere in the search path, with a `manifest.path` definition. However, I'm having some problems reproducing it on my environment - presumably west is finding a config file somewhere but I can't find it. I can try a bit harder to reproduce if you don't see this issue.

mbolivar, 2019-06-05:
> @mbolivar I think there's a bit of a bootstrapping problem for non-Zephyr projects when you don't already have a west config file defined somewhere in the search path, with a manifest.path [...]

I think the "trick" here is that `west init` has a default manifest repository URL, which happens to point at the zephyr repo. You can `west init -m git@github.com:you/your-non-zephyr-repo`, and west will happily point `manifest.path` at `your-non-zephyr-repo` in the local config file.

So if I'm not misunderstanding you, I think there's no issue with that.

bengartner, 2019-06-06:
> So if I'm not misunderstanding you, I think there's no issue with that.

No. :) I was doing `west init -m <repo>` and had some issues doing a clean checkout that seemed to be resolved after I created my own minimal config file. But I'm on the tip of master now and I haven't been able to reproduce the issue so I'm going to consider it solved/imaginary unless I see it again.

marc-hb, 2020-02-14:
BTW west documentation is not even a first level item in https://docs.zephyrproject.org/2.1.0/guides/index.html

This really doesn't make it look like something that can be used outside Zephyr.

mbolivar, 2020-02-15:
The idea would be to host its documentation on readthedocs instead. But that will take quite some doing. We want the code to be ready before we try to do that.