zephyrproject-rtos/west #251: Auto-detect board from existing build folder

carlescufi, 2019-04-25:
`west build -b reel_board samples/hello_world`
and then
`west build samples/philosophers`
should auto-detect the board and build philosophers for the reel_board

mbolivar, 2019-04-30:
I'm not sure about this one. A build directory is specific to a board and sample application. Switching them would require making pristine at the very least. I'm concerned about west build getting too smart for its own good in terms of DWIM when things mismatch.

I'd rather add an option setting a default board. I have a patch for this that I'll rebase onto your latest west boards changes.

mbolivar, 2019-05-15:
@carlescufi I think that between `build.pristine`, `--pristine`, `build.board`, and tab completion of board names, the pain points here are likely gone; can we close this?

carlescufi, 2019-05-16:
Agreed, closing

closed 2019-05-16 by carlescufi