zephyrproject-rtos/west #252: Add support for Make as a CMake generator

carlescufi, 2019-04-26:
Currently the code uses `DEFAULT_CMAKE_GENERATOR = Ninja`. Add a command-line switch to use Make instead.

mbolivar, 2019-04-26:
does `-- -G'Unix Makefiles'` not work?

carlescufi, 2019-04-26:
@mbolivar it works indeed, but it's a bit cumbersome. I wonder if we could give special treatment to the generator and add a `-g/--generator` with shortened names:

```
-g ninja
-g make
```

carlescufi, 2019-04-27:
In addition to a `-g` command-line option, I suggest we add a config option:

```
[build]
generator = ninja/make/...
```

galak, 2019-04-29:
We should at least have a way of setting the preference in the config file.

mbolivar, 2019-04-30:
A `build.generator` config option seems fine to me. I'm -1 on wrapping cmake with west options unnecessarily though. That way leads madness.

mbolivar, 2019-04-30:
> generator = ninja/make/...

We should use the actual generator names supported by CMake and pass them unmodified (so 'Unix Makefiles', 'Ninja', etc.). Let cmake do the validation for us, etc.

closed 2019-05-06 by nashif