zephyrproject-rtos/west #253: Add a command-line option for verbose Makefiles

carlescufi, 2019-04-26:
Add a command-line switch that adds `-DCMAKE_VERBOSE_MAKEFILE:BOOL=ON` to the CMake invocation.

mbolivar, 2019-04-26:
`-- -DCMAKE_VERBOSE_MAKEFILE=YES`

carlescufi, 2019-04-26:
@mbolivar I would suggest we automatically add that option when using `-vv` so that:

- `-v` shows the cmake/ninja command
- `-vv` shows the actual commands executed by ninja

mbolivar, 2019-04-27:
Doing so would force a CMake re-run, which may not be what users want.

mbolivar, 2019-05-01:
Is adding the CMake variable setting to the command line really so bad? Can we close this?

galak, 2019-05-01:
> Is adding the CMake variable setting to the command line really so bad? Can we close this?

So while not exactly related to this, I feel like having this feature allows one to replicate the following usage model:

```
$ cmake -DBOARD=frdm_k64f ..
$ make VERBOSE=1
```

mbolivar, 2019-05-02:
> So while not exactly related to this, I feel like having this feature allows one to replicate the following usage model:

OK, so how about the ability to pass additional options to the build tool?

```
west build --build-opt "VERBOSE=1" -- -G'Unix Makefiles'
```

galak, 2019-05-02:
I'm good with adding a --build-opt and just having the other passed to through to cmake directly.  Might be nice to add something in the west docs about `-DCMAKE_VERBOSE_MAKEFILE:BOOL=ON`

mbolivar, 2019-05-04:
> I'm good with adding a --build-opt and just having the other passed to through to cmake directly. Might be nice to add something in the west docs about -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON

I will do both things in https://github.com/zephyrproject-rtos/zephyr/pull/15801 before dropping the DNM.

closed 2019-05-06 by nashif