zephyrproject-rtos/west #254: Show the CMake and Ninja/make command-line strings

carlescufi, 2019-04-26:
When building with -v, show the actual commands for cmake and ninja.

mbolivar, 2019-04-26:
already done, you just need `-vv`:

https://github.com/zephyrproject-rtos/west/blob/master/src/west/cmake.py#L33

 feel free to lower the level of verbosity

galak, 2019-04-26:
>already done, you just need `-vv`: ...

@mbolivar 

Would be nice if we could be consistent here.  Right now for ‘flash’ only a single ‘-v’ is needed to see the command.

carlescufi, 2019-04-26:
@galak I have a PR that fixes this.

mbolivar, 2019-04-30:
> Would be nice if we could be consistent here. Right now for ‘flash’ only a single ‘-v’ is needed to see the command.

Makes sense to me, will wait for PR from @carlescufi

closed 2019-05-06 by nashif